BehaviorTree/BehaviorTree.CPP #640: colcon compilation failed

tony-p, 2023-08-18:
On Ubuntu I am getting compilation warnings/errors, many are due to minitrace, and with the latest g++ version also some in `simple_string.hpp`

What is the known working compiler verison? I have tried g++ 13.1, 12.3, and 11.4. 

With 11.4 all issues and warnings are in minitrace

tony-p, 2023-08-18:
Also this one which is actually an error

```
BehaviorTree.CPP/3rdparty/minitrace/minitrace.cpp:347:112: error: expected ')' before 'PRId64'

  347 | linebuf), "%s{\"cat\":\"%s\",\"pid\":%i,\"tid\":%i,\"ts\":%" PRId64 ",\"ph\":\"%c\",\"name\":\"%s\",\"args\":{%s}%s}",

```

but that seems compiler version related and that is why I was wondering which compiler (version) you use so I can actually build it on ubuntu.

https://github.com/pytorch/pytorch/issues/5217

facontidavide, 2023-08-28:
I can confirm that the library compiles in both Ubuntu 22.04 and Ubuntu 20.04

tony-p, 2023-08-29:
It seems to be an issue of building inside a conda workspace. The standard macros don't seem to be available by default in a linux conda environment (windows doesn't have the same issue, but is a different build chain).

I found a solution used by others that adds the following to includes of `minitrace.cpp` which solves the issue for me
```cpp
#define __STDC_FORMAT_MACROS
#include <inttypes.h>
```

facontidavide, 2023-08-31:
great, can you submit a PR?

tony-p, 2023-08-31:
I've been working on a general conda support PR (this branch https://github.com/tony-p/BehaviorTree.CPP/tree/feature/conda-support) 

I think it is basically working for windows. The linux portion still need to fix the last issues. The main issue is finding an including of the  zeromq library that works both with conda, and the current build methods.

Once that is achieved, it should also be trivial to achieve this idea https://github.com/BehaviorTree/BehaviorTree.CPP/discussions/644

tony-p, 2023-09-01:
Created this PR https://github.com/BehaviorTree/BehaviorTree.CPP/pull/650

closed 2023-09-06 by facontidavide