BehaviorTree/BehaviorTree.CPP #649: Need Complete Documentation for all BT Nodes with Examples

vignesh-katomaran, 2023-08-30:
currently unable to find the documentation for some of the nodes also it will be better if we have complete documentation for all nodes and multiple usage examples.

facontidavide, 2023-09-06:
The best option is to read the comments in the header files, most of the time.

Any specific list of nodes I should prioritize, in your opinion?

dt1729, 2023-09-21:
I don't know who will be helped by this but these linked helped me a lot in the past:
[Doxygen Documentation](http://docs.ros.org/en/noetic/api/behaviortree_cpp/html/classBT_1_1TreeNode.html)
[fkie_Behaviortree repo's wiki](https://github.com/fkie/fkie_behavior_trees/wiki/5.-How-to-debug-BTs-in-ROS)

vignesh-katomaran, 2023-10-04:
@facontidavide Sequence Star

sven-hoek, 2025-05-08:
For anyone reaching this in search of the docs, here are links to versions of other ROS2 distros.

ROS2 Humble: [https://docs.ros.org/en/humble/p/behaviortree_cpp/](https://docs.ros.org/en/humble/p/behaviortree_cpp/)
ROS2 Rolling: [https://docs.ros.org/en/rolling/p/behaviortree_cpp/](https://docs.ros.org/en/rolling/p/behaviortree_cpp/)

Not sure, which BT.CPP version each one uses or if there's a documentation build published for this lib outside the ROS context. If it doesn't exist yet, it could be well worth creating one, e.g. automatically created via GH Action and published as GH Page - especially as the website doesn't seem to list all Nodes that come with the library, which is fair enough as it seems to be a place for a handcrafted description of each node's behavior and takes time to create.

If get some time to spend, I'd see how to create such a GH page from the code documentation but I don't think I will anytime soon.

sven-hoek, 2025-05-08:
Also, I missed that there already is a Doxyfile, so anyone who has Doxygen installed can simply clone the repo/checkout the relevant commit/branch and run `doxygen Doxyfile` inside to generate the documentation locally (just open `doc/index.html` afterwards)

sven-hoek, 2025-05-08:
Actually, it was easier than thought, as Doxygen was already set up. PR is up: https://github.com/BehaviorTree/BehaviorTree.CPP/pull/972