BehaviorTree/BehaviorTree.CPP #651: Multiple memory/threading errors under sanitizers.

melg8, 2023-09-05:
# Problem
When you compile project (at least) on linux with flags:
1. -fsanitize=address 
2. -fsanitize=thread 
multiple tests from unit test suit fail with error reports from sanitizers.

# How to reproduce.
1. Add next line to main CMakeLists.txt
``` 
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -fsanitize=address")
```
2. Compile project with tests enabled.
3. Run tests suite.
4. Observe [errors](https://github.com/melg8/BehaviorTree.CPP/actions/runs/6081036796/job/16495942878#step:10:242) - example from CI run of my fork. 

# Notice
1. For address sanitizer it seems errors occur only when certain tests are run together. For example i found pair:
 CoroTest.do_action + BehaviorTreeFactory.WrongTreeName execution will enable WrongTreeName test failure with stack overflow error.
 From that pair i tried to make minimal reproducible test. It looks like this so far:
 ```
 void TestLocal(const int32_t root_blackboard[4],
                                std::string main_tree_ID)
{
  static_assert(sizeof(main_tree_ID) == 32);
  std::string output_tree;
  throw 42;
}

void TestWrapper(const std::string& tree_name,
                int32_t blackboard [4] = {}) {
  TestLocal(blackboard, tree_name);
}

TEST(CoroTest, do_action)
{
  BT::NodeConfig node_config_;
  node_config_.blackboard = BT::Blackboard::create();
  BT::assignDefaultRemapping<SimpleCoroAction>(node_config_);
  SimpleCoroAction node(milliseconds(1), false, "Action", node_config_);
  node.executeTick();

  try {
    TestWrapper("Wrong Name");
  } catch (...) {

  }
}
```
I tried to remove all functions/parameters/types which doesn't affect error. 
I'm not very familiar with this library, but i suspect that "address" issues are connected to coro library usage with exceptions thrown after that interaction which enables stack buffer overflow error.

-fsanitize=thread showing different places with multiple issues. I didn't try to look to much into it.

# Expected
You can checkout info about sanitizers [here](https://github.com/google/sanitizers) they are integrated into most of modern c++ compilers and have quite good signal/noise ratio, so it would be preferable for any significant library to have test suit cleanly passing without any errors under sanitizers. And if any of them have false positives - they also can be added to exclusion file so tests still pass cleanly. Also its great idea to add sanitizers into CI run to ensure clean runs in future.

facontidavide, 2023-09-05:
Thanks, I will address this this week, if I can

facontidavide, 2023-09-06:
I investigated and the warning in the thread sanitizer are a well known false positive with condition variables.

The warning with the address sanitizer seems also a false positive, due to the fact that gtest was not compiled with the sanitizer.

closed 2023-09-06 by facontidavide

melg8, 2023-09-07:
@facontidavide thanks for fast response.

Although address sanitizer indeed maybe reporting false positives, even in that case. I don't think gtest is the reason. I implemented standalone example from 2 of your tests which reproduce error under sanitizers while not being linked to gtest at all.
Source code: 
``` C++
// Changes start.

#include <iostream>

#define EXPECT_EQ(expected, actual) \
  if (expected != actual) { \
    std::cout << "Not equal, returning 121" << std::endl; \
    exit(121); \
  }

#define EXPECT_FALSE(condition) \
  if (condition) { \
    std::cout << "Condition is not false, returning 122" << std::endl; \
    exit(122); \
  } \

#define TEST(name, subname) \
void name##_##subname()

#define EXPECT_THROW(expr, exception_type) \
  try \
  { \
    expr; \
    std::cout << "No exception was thrown, returning 123" << std::endl; \
    exit(123); \
  } catch (const exception_type &) { \
    std::cout << "Expected exception was thrown" << std::endl; \
  } catch (...) { \
    std::cout << "Unexpected exception was thrown, returning 124" << std::endl; \
    exit(123); \
  }

// Changes end.

// gtest_coroutines.cpp start.
#include "behaviortree_cpp/decorators/timeout_node.h"
#include "behaviortree_cpp/behavior_tree.h"
#include <chrono>
#include <future>

using namespace std::chrono;

using Millisecond = std::chrono::milliseconds;
using Timepoint = std::chrono::time_point<std::chrono::steady_clock>;

class SimpleCoroAction : public BT::CoroActionNode
{
public:
  SimpleCoroAction(milliseconds timeout, bool will_fail, const std::string& node_name,
                   const BT::NodeConfig& config) :
    BT::CoroActionNode(node_name, config),
    will_fail_(will_fail),
    timeout_(timeout),
    start_time_(Timepoint::min())
  {}

  virtual void halt() override
  {
    std::cout << "Action was halted. doing cleanup here" << std::endl;
    start_time_ = Timepoint::min();
    halted_ = true;
    BT::CoroActionNode::halt();
  }

  bool wasHalted()
[... 167 lines cut ...]
  Right alloca redzone:    cb
==22666==ABORTING
```

While ldd shows this dependencies:
```
ldd ./examples/address_sanitizer_standalone 
        linux-vdso.so.1 (0x00007ffeaa946000)
        libasan.so.8 => /usr/lib/libasan.so.8 (0x00007fc140a4f000)
        libbehaviortree_cpp.so => /home/user/work/BehaviorTree.CPP/build/libbehaviortree_cpp.so (0x00007fc140591000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007fc14036b000)
        libm.so.6 => /usr/lib/libm.so.6 (0x00007fc14028b000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007fc14026a000)
        libc.so.6 => /usr/lib/libc.so.6 (0x00007fc140082000)
        libdl.so.2 => /usr/lib/libdl.so.2 (0x00007fc14007d000)
        librt.so.1 => /usr/lib/librt.so.1 (0x00007fc140078000)
        libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007fc140073000)
        libzmq.so.5 => /nix/store/5nbd10268965s5v3jwx6p2m0gh9wca57-zeromq-4.3.4/lib/libzmq.so.5 (0x00007fc13ff8a000)
        libsqlite3.so.0 => /nix/store/n12a68qch9s85k6ni4m4r4xxr8lwys1i-sqlite-3.41.2/lib/libsqlite3.so.0 (0x00007fc13fe39000)
        /nix/store/46m4xx889wlhsdj72j38fnlyyvvvvbyb-glibc-2.37-8/lib/ld-linux-x86-64.so.2 => /nix/store/46m4xx889wlhsdj72j38fnlyyvvvvbyb-glibc-2.37-8/lib64/ld-linux-x86-64.so.2 (0x00007fc1410f5000)
        libsodium.so.23 => /nix/store/b19c558b7c21a72hxr6cq61jhzz3h8hq-libsodium-1.0.18/lib/libsodium.so.23 (0x00007fc13fdde000)
        libz.so.1 => /usr/lib/libz.so.1 (0x00007fc13fdbf000)
```

I would suggest to further investigate this case and not dismiss it only as false positive from gtest influence.