BehaviorTree/BehaviorTree.CPP #652: [Bug!!!] Reactive sequence does not follow the expected logic

xiaofyao, 2023-09-06:
BT structure:

```shell
----------------
ReactiveSequence
   Node1
   Node2
----------------
```

Node1 returns success and Node2 returns running. 
If the software follows the expected logic of the reactive sequence control node, it will trigger Node1 again every time, but Node1 is only triggered once for the first time, and Node2 is still triggered every time after that. This does not match the official description of the reactive sequence control node.

I can't debug the behavior tree dynamic library using GDB, please help me solve this problem. And could you also tell me how to debug the behavior tree dynamic library using GDB.

Thanks!

xiaofyao, 2023-09-06:
software version: v3.8

facontidavide, 2023-09-06:
Please provide an example or unit test to reproduce the issue

xiaofyao, 2023-09-06:
The code is as follows:

```c++
#include <unistd.h>

#include "behaviortree_cpp_v3/bt_factory.h"
#include "behaviortree_cpp_v3/loggers/bt_zmq_publisher.h"

using namespace BT;

namespace chr = std::chrono;

class Node1 : public BT::StatefulActionNode {
 public:
  // Any TreeNode with ports must have a constructor with this signature
  Node1(const std::string& name, const BT::NodeConfiguration& config) : StatefulActionNode(name, config) {}
  static BT::PortsList providedPorts() { return {}; }
  // this function is invoked once at the beginning.
  BT::NodeStatus onStart() override {
    _completion_time = chr::system_clock::now() + chr::milliseconds(2200);
    return BT::NodeStatus::RUNNING;
  }

  // If onStart() returned RUNNING, we will keep calling
  // this method until it return something different from RUNNING
  BT::NodeStatus onRunning() override {
    if (chr::system_clock::now() >= _completion_time) {
      std::cout << "[ Node1: FINISHED ]" << std::endl;
      return BT::NodeStatus::SUCCESS;
    }
    return BT::NodeStatus::RUNNING;
  }

  // callback to execute if the action was aborted by another node
  void onHalted() override {}

 private:
  chr::system_clock::time_point _completion_time;
};

class Node_2 : public BT::StatefulActionNode {
 public:
  // Any TreeNode with ports must have a constructor with this signature
  Node_2(const std::string& name, const BT::NodeConfiguration& config) : StatefulActionNode(name, config) {}
  static BT::PortsList providedPorts() { return {}; }
  // this function is invoked once at the beginning.
  BT::NodeStatus onStart() override {
    _completion_time = chr::system_clock::now() + chr::milliseconds(2200);
    return BT::NodeStatus::RUNNING;
  }

  // If onStart() returned RUNNING, we will keep calling
  // this method until it return something different from RUNNING
  BT::NodeStatus onRunning() override {
    if (chr::system_clock::now() >= _completion_time) {
      //std::cout << "[ Node_2: FINISHED ]" << std::endl;
      //return BT::NodeStatus::SUCCESS;
    }
    return BT::NodeStatus::RUNNING;
  }

  // callback to execute if the action was aborted by another node
  void onHalted() override {}

 private:
  chr::system_clock::time_point _completion_time;
};

int main() {
  // We use the BehaviorTreeFactory to register our custom nodes
  BehaviorTreeFactory factory;
  // The recommended way to create a Node is through inheritance.
  factory.registerNodeType<Node_1>("Node1");
  factory.registerNodeType<Node_2>("Node2");

  auto tree = factory.createTreeFromFile("../my_demo_0.xml");

  BT::PublisherZMQ publisher(tree);

  printTreeRecursively(tree.rootNode());

  //tree.tickRootWhileRunning();

  while (1) {
    // To "execute" a Tree you need to "tick" it.
    // The tick is propagated to the children based on the logic of the tree.
    // In this case, the entire sequence is executed, because all the children
    // of the Sequence return SUCCESS.
    tree.tickRoot();

    tree.sleep(std::chrono::milliseconds(100));
  }

  return 0;
}
```

The behavior tree XML file is as follows:
`my_demo_0.xml`
```xml
<?xml version="1.0" encoding="UTF-8"?>
<root BTCPP_format="3"
      main_tree_to_execute="my_line_fly_tree">
  <BehaviorTree ID="my_line_fly_tree">
    <ReactiveSequence>
      <Node1/>
      <Node2/>
    </ReactiveSequence>
  </BehaviorTree>

  <!-- Description of Node Models (used by Groot) -->
  <TreeNodesModel>
    <Action ID="Node1"
            editable="true"/>
    <Action ID="Node2"
            editable="true"/>
  </TreeNodesModel>

</root>
```

@facontidavide

facontidavide, 2023-09-06:
Mistery solved...

**ReactiveSequence** can have only a single child returning RUNNING.

**Node1** must be synchronous, can not be asynchornous

facontidavide, 2023-09-06:
> could you also tell me how to debug the behavior tree dynamic library using GDB.

As any other software, no difference; it is outside my scope as maintainer to teach you GDB :smiling_face_with_tear: 

Consider using logging, for easy debugging.

closed 2023-09-06 by facontidavide

xiaofyao, 2023-09-06:
Thank you for your answer, which solved my big problem! Could you introduce the use of these nodes more fully in the official documents, so that we can use the behavior tree more conveniently. @facontidavide

msmcconnell, 2023-12-01:
@facontidavide Appreciate you adding the error log, but I don't see why this is an acceptable solution to the stated issue. A ReactiveSequence is a description of the ticking order and in my opinion should be independent from the Synchronous or ASynchronous nature of the underlying nodes. In fact using an ASynchronous nodes is the exact example given in the docs https://www.behaviortree.dev/docs/tutorial-basics/tutorial_04_sequence (though it's just one in that case) so I think users might expect it to be supported. 

I would think that if an earlier node became running in the ReativeSequence the correct behavior would be to halt the following children. Logically that makes sense as the sequence can no longer confirm that it's appropriate to run the following children. Then when the prior child returns success it restarts the following children. I had a tree setup for path planning that worked in 4.0.4 based on this assumption but broke when I upgraded to the newer version. It looked like this

[image]

In this case, there is a precondition check PathIsAvailable in-front of an ASynchronous path planning node. These are followed by a ASynchronous path following node. The flow is basically, check if a path of a given type is available, if not then plan it. Else follow the planned path. It's possible at runtime for PathIsAvailable to become false (FAILURE) when we want to replan. In this case the PathPlan action gets called again, but this now causes an error since the FollowPath is still running. This structure can certainly be reworked to avoid this issue but it's not clear to me that there's a strong reason it shouldn't be supported already (and indeed it did use to work).

facontidavide, 2023-12-02:
In general, wrapping multiple async nodes into a Sequence is what you want to do