BehaviorTree/BehaviorTree.CPP #653: SetBlackboard crashing when used with a typed remapped port

afrixs, 2023-09-06:
**Version**: 3.8.5

Custom node (the important part is that `<bool>` is used inside `providedPorts()`):
```
#include "behaviortree_cpp_v3/behavior_tree.h"

namespace bt_test
{
class Assert : public BT::SyncActionNode
{
  public:
    Assert(const std::string& name, const BT::NodeConfiguration& config)
      : BT::SyncActionNode(name, config) {}

    static BT::PortsList providedPorts()
    {
        return {BT::InputPort<bool>("condition")};
    }

  private:
    virtual BT::NodeStatus tick() override {
        if (getInput<bool>("condition").value())
            return BT::NodeStatus::SUCCESS;
        else
            return BT::NodeStatus::FAILURE;
    }
};

} // end namespace
```
BT:
```
<?xml version='1.0'?>
<root main_tree_to_execute = "MainTree" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <BehaviorTree ID="MainTree">
    <Sequence>
      <SubTreePlus ID="Init" test="{test}" />
      <Assert condition="{test}" />
    </Sequence>
  </BehaviorTree>

  <BehaviorTree ID="Init">
    <SetBlackboard output_key="test" value="true"/>
  </BehaviorTree>
</root>
```

**Expected behavior**:
SUCCESS

**Actual behavior**:
```
terminate called after throwing an instance of 'BT::LogicError'
  what():  Blackboard: once declared, the type of a port shall not change. Previously declared type [bool] != new type [std::string]
```

**Notes**:
* The crash doesn't occur when `<SetBlackboard output_key="test" value="true"/>` is used directly in the MainTree
* Same happens with `SubTree`

facontidavide, 2023-09-06:
thanks for reporting. Version 4.x on master is also affected. I fixed it there, but I will backport the fix to 3.8

facontidavide, 2023-09-06:
Fixed on branch v3.8 too

facontidavide, 2023-09-06:
thanks for providing a way to test it :smiley:

closed 2023-09-06 by facontidavide

bektaskemal, 2023-10-24:
Hi, will there be a 3.8 release for this?

facontidavide, 2023-10-24:
@bektaskemal when you ask about a release, which ROS distro are you talking about? I think it was released