BehaviorTree/BehaviorTree.CPP #655: Blackboard Ports Documentation Clarification

jmount-boss, 2023-09-08:
[Tutorial 02 Blackboard and Ports](https://www.behaviortree.dev/docs/tutorial-basics/tutorial_02_basic_ports) describes how one can pass data in to and out from nodes. However, I think some of the naming/semantics could be clarified to improve it and reduce confusion. 

I believe adding something akin to the following elements within the tutorial would help.

> Terminology:
> - **Blackboard Key**: The string identifier used to access an entry within the Blackboard.
> - **Blackboard Value**: The value of an entry within the Blackboard for a given Blackboard Key.
> - **Port Name**: The string identifier used to define the port. This is the name of the XML attribute and is set within a node's ports list.
> - **Port Value**:  The value given to the port within the XML code. 

I think this would be good at the start of the tutorial, and then use these terms (or whatever is chosen) throughout. 

> A Generic Port Example:
> ```c++
> static PortsList MyBTNode::providedPorts()
> {
>      return { InputPort<Type>("PORT_NAME") };  
>      // the Port Value will be set within XML using the PORT_NAME
>      // see the following XML snippet
>     
>      // or for an Output Port
>      // return { OutputPort<Type>("PORT_NAME") };
> }
> ```
> ```xml
> // Set the Port Value using the Port Name within XML
> <MyBTNode PORT_NAME="PORT_VALUE"/> 
> ```
Probably best after the Input and Output Port examples, but before *A Complete Example*.

Even with these examples I would still have the following questions:

- Why, under the *A Complete Example* in XML line describing the ThinkWhatToSay node `<ThinkWhatToSay   text="{the_answer}"/>`, do we need to wrap `the_answer` in braces? It is an output port, so wouldn't the value of `text` always be a Blackboard Key?
- How would one "hard-code" the Blackboard key used by an Output Port? Obviously, this wouldn't lead to good code reuse, but this is a hypothetical question.

So maybe adding a note or two at the end would be beneficial, something like:
> **Notes**
> - When setting the Port Value to be a Blackboard Key, within your XML, it always needs to be wrapped in '{}' regardless if it's an Output or Input port. 

It should also be stated that, obviously, I am learning the framework so my suggestions may be wrong, but the key principle behind them is still valid. That is to say, there is some confusion here that could be overcome with minimal additions.

facontidavide, 2023-09-08:
Great suggestions. I will

jmount-boss, 2023-09-08:
Awesome! The tutorials are great in general though. And a great library.

closed 2024-05-08 by facontidavide