BehaviorTree/BehaviorTree.CPP #660: `convertFromString` to scoped enum is not invoked when passing string literal to the input port

alsora, 2023-09-15:
I'm trying to use the `convertFromString` function to convert a string into an enum value.
This works if the string value is passed to the input port as a blackboard variable, but it doesn't work (i.e. the  `convertFromString` function is not invoked) when using a string literal.

Consider the following C++ implementation

```
enum class MyValue
{
    ALPHA,
    BETA
};

namespace BT
{
    template <> inline SortFrontiersHeuristic convertFromString(StringView str)
    {
        MyValue v;
        if (std::string(str.data()) == "ALPHA") {
            v = SortFrontiersHeuristic::DISTANCE;
        } else if (std::string(str.data()) == "BETA") {
            v = SortFrontiersHeuristic::SIZE;
        } else {
            throw BT::RuntimeError(BT::StrCat("Can't convert string [", str, "] to value"));
        }
        return v;
    }

class MyNode: public SyncActionNode
{
  public:
    MyNode(const std::string& name, const NodeConfig& config):
      SyncActionNode(name,config)
    {}

    static PortsList providedPorts()
    {
      return { InputPort<MyValue>("in_value") };
    }

    NodeStatus tick() override
    {
     auto v = getInput<MyValue>("in_value");
     switch (v.value()) {
     case MyValue::ALPHA:
         std::cout<<"Value was alpha << std::endl;
         break;
     case MyValue::BETA:
         std::cout<<"Value was beta << std::endl;
         break;
     }
      return NodeStatus::SUCCESS;
    }
};
} // end namespace BT
```

This works as expected (i.e. the node prints "ALPHA")
```
<Script code=" my_value := 'ALPHA' "/>
<Action ID="MyNode" in_value="{my_value}"  />
```

This doesn't work (the node prints "BETA", likely because it interprets the string literal as "non-zero"

```
<Action ID="MyNode" in_value="ALPHA"  />
```

facontidavide, 2023-09-15:
yes, this is a known limitation. I will check if this can be corrected

alsora, 2023-09-15:
Thank you!
Is this limitation caused by the custom type being an enum or does it apply to all custom types?

facontidavide, 2023-10-16:
I am trying to create  unit test of the code you mention here, but this doesn't make much sense:

```c++
template <> inline SortFrontiersHeuristic convertFromString(StringView str)
    {
        MyValue v;
        if (std::string(str.data()) == "ALPHA") {
            v = SortFrontiersHeuristic::DISTANCE;
        } else if (std::string(str.data()) == "BETA") {
            v = SortFrontiersHeuristic::SIZE;
        } else {
            throw BT::RuntimeError(BT::StrCat("Can't convert string [", str, "] to value"));
        }
        return v;
    }
```

You are trying to write  `SortFrontiersHeuristic` into `MyValue`, that won't compile

facontidavide, 2023-10-16:
I don't see WHY your `SortFrontiersHeuristic convertFromString(StringView str)` should be invoked (it shouldn't!!!!)

The port is declared as type  `MyValue`, not `SortFrontiersHeuristic`, therefore the template specialization of the latter will never be invoked

facontidavide, 2023-10-16:
Another error in your code is here:

```xml
<Script code=" my_value := 'ALPHA' "/>
```

You are EXPLICITLY saying that it is a string. That would be equivalent to C++

```cpp
MyValue v = "ALPHA";
```

 It should be instead
```xml
<Script code=" my_value := ALPHA "/>
```

closed 2023-10-16 by facontidavide

alsora, 2024-05-16:
Hi, sorry for disappearing from this issue for months.

> You are trying to write SortFrontiersHeuristic into MyValue, that won't compile
The code snippet that I posted was a simplified version of a real BT node from my app, and apparently I made a mistake with a "replace all". Treat `SortFrontiersHeuristic` and `MyValue` as the same thing.

The root of my problems was a lack of `factory.registerScriptingEnums<MyValue>();`.

Without it, there's indeed an inconsistent behavior when using input ports with enum class as types.

This will call into a user-defined `convertFromString(StringView str)`
```
<Script code=" my_value := 'ALPHA' "/>
<Action ID="MyNode" in_value="{my_value}"  />
```

This will not call into it, and always treat the value as "1" (I assume because the string is not empty).
The `convertFromString(StringView str)` won't be called

```
<Action ID="MyNode" in_value="ALPHA"  />
```

If your custom types is not an enum, the behavior on the other hand is consistent in the two cases.

alsora, 2024-05-16:
Important correction.
The reason why I was getting "1" as value in the wrong examples was because in my code I was treating that as a default value (I had wrapped the `getInput` and I was ignoring the error state in the `nonstd::expected` when the user provided a default value).

Probably worth a separate discussion, but it would be nice to allow to programmatically distinguish different error types.
Right now the code uses `using Expected = nonstd::expected<T, std::string>;`.
The string could be replaced/augmented with an error code.

For example in my use case I want to distinguish when a `getInput` fails because there's no input value or when there's a developer mistake (type mismatch or trying to retrieve values from a non-existent port, etc)