BehaviorTree/BehaviorTree.CPP #679: Library support for emsdk (WebAssembly)

coder137, 2023-10-25:
Was looking into creating GDExtension bindings with the Godot engine using this library

Reference (https://github.com/godotengine/godot-cpp)
Different platforms that should be supported (https://github.com/godotengine/godot-cpp/blob/master/test/project/example.gdextension)

When trying to compile with emsdk (https://emscripten.org/index.html) for a Web release I get compile errors.

By looking at the CMake scripts have removed the `shared_library_*` files from compilation as well as removed the `SharedLibrary` usage from the `BehaviorTreeFactory::registerFromPlugin` API.
According to me this was the only OS specific dependency of this project.

Is there any other change required to make WebAssembly builds compile?

Current scons script
```python
Import("env")

BEHAVIORTREE_FOLDER = "BehaviorTree.CPP"
env.Append(CPPPATH=[
    f"{BEHAVIORTREE_FOLDER}/include/",
    f"{BEHAVIORTREE_FOLDER}/3rdparty/",
    f"{BEHAVIORTREE_FOLDER}/3rdparty/lexy/include/"
])
env.Append(CPPDEFINES=[
    "BTCPP_LIBRARY_VERSION=\\\"4.4.0\\\"",
    "LEXY_HAS_UNICODE_DATABASE=1"
])
sources = [
    f"{BEHAVIORTREE_FOLDER}/src/action_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/basic_types.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/behavior_tree.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/blackboard.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/bt_factory.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/decorator_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/condition_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/control_node.cpp",
    # f"{BEHAVIORTREE_FOLDER}/src/shared_library.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/tree_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/script_parser.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/json_export.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/xml_parsing.cpp",

    f"{BEHAVIORTREE_FOLDER}/src/actions/test_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/actions/sleep_node.cpp",

    f"{BEHAVIORTREE_FOLDER}/src/decorators/delay_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/decorators/inverter_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/decorators/repeat_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/decorators/retry_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/decorators/timeout_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/decorators/subtree_node.cpp",

    f"{BEHAVIORTREE_FOLDER}/src/controls/if_then_else_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/controls/fallback_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/controls/parallel_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/controls/parallel_all_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/controls/reactive_sequence.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/controls/reactive_fallback.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/controls/sequence_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/controls/sequence_star_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/controls/switch_node.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/controls/while_do_else_node.cpp",

    f"{BEHAVIORTREE_FOLDER}/src/loggers/bt_cout_logger.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/loggers/bt_file_logger_v2.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/loggers/bt_minitrace_logger.cpp",
    f"{BEHAVIORTREE_FOLDER}/src/loggers/bt_observer.cpp",

    f"{BEHAVIORTREE_FOLDER}/3rdparty/tinyxml2/tinyxml2.cpp",
    f"{BEHAVIORTREE_FOLDER}/3rdparty/minitrace/minitrace.cpp",
]

# # TODO, Might need to check for WASM applications
# if env['PLATFORM'] == 'windows':
#     sources.append(f"{BEHAVIORTREE_FOLDER}/src/shared_library_WIN.cpp")
# else:
#     sources.append(f"{BEHAVIORTREE_FOLDER}/src/shared_library_UNIX.cpp")

print(f"Sources length: {len(sources)}")
libpath = env.File(
    f"bin/{env['platform']}/libbehaviortree_cpp{env['suffix']}{env['LIBSUFFIX']}")
library = env.StaticLibrary(
    target=libpath, source=sources)
print(library)

env.Default(library)
env.AppendUnique(LIBS=[libpath])
```

coder137, 2023-10-27:
To get the library to compile with emscripten (https://emscripten.org/index.html), Made the following changes

## Removed `BT::SharedLibrary` from `registerFromPlugin`

- This is an OS-specific feature that can be used only for WIN and UNIX systems

## Removed `minicoro.h` usage

- Was giving error `templates must have C++ linkage`. The minicoro.h library uses `extern "C"`, should use `extern "C++"` instead.
- This resulted in removing the entire `CoroActionNode` from the library

## Deleted `std::from_chars` API when using emscripten

- In API `Any::stringToNumber` in `safe_any.hpp`, 
changed 
`#if __has_include(<GL/gl.h>)` to
`#if __has_include(<GL/gl.h>) && !defined(__EMSCRIPTEN__)`

This library now compiles with the emscripten (em++) compiler. 

According to me, For 

- 1st change can be 
  - Compiled out during the build system stage (files are not compiled and linked)
  - Under a compile flag like `if defined(__EMSCRIPTEN__)` to conditionally remove these features.
- 2nd change could 
  -  conditionally remove this feature as well using `if defined(__EMSCRIPTEN__)`
  - require a fix using extern "C++" instead of extern "C" to remove compile errors (unsure how to proceed with this)
- Unsure about the impact of the 3rd change. Maybe there is a cleaner way to handle this conversion

@facontidavide Please let me know what you think. 
I can make these changes for Webassembly compilation via a PR.

facontidavide, 2023-10-27:
I am happy to accept a PR as long as all the changes are wrapped into an` if defined(__EMSCRIPTEN__)` section and nothing changes when compiling with a regular compiler

facontidavide, 2024-02-07:
Any update on this?

coder137, 2024-02-10:
> Any update on this?

Had forgotten about this. Thanks for pinging.
Have made a PR where emsdk 3.1.47 can compile BehaviorTree.CPP as a static lib.