BehaviorTree/BehaviorTree.CPP #681: Can't read custom type ports

siddux, 2023-10-25:
Hi, 

I am trying to get the value of a custom type port but I'm failing. I have a custom struct similar to this one:

```bash
struct MyStruct 
{ 
  string a;
  int b; 
};
```

I set the value into the blackboard as follows:

```bash
MyStruct my_struct;
my_struct.a = "Hello world!";
my_struct.b = 42;
blackboard_->set("my_struct_key", my_struct);
```

And then in a node I try to get this value as:

```bash
MyStruct my_struct;
getInput<MyStruct>("my_struct_key", my_struct);
printf("My struct is saying %s %i times", my_struct.a.c_str(), my_struct.b);
```
And I get weird characters from this.

I tried the same but using built-in types like string or int and it works so I should be missing something. I already tried to implement the `template <> inline MyStruct convertFromString(StringView str)` function but I get the same result. Any working example on something similar? 

Thanks!

facontidavide, 2023-10-25:
Please send a full example that I can compile and execute myself

siddux, 2023-10-25:
Preparing a minimum example I found the error (and it wasn't related to the BT library). Anyway, thanks for your quick response!

closed 2023-10-25 by siddux