BehaviorTree/BehaviorTree.CPP #683: Compilation error:  Accessing Protected config() Method in bt_flatbuffer_helper.h

raviBhadeshiya, 2023-10-26:
Encountered compilation errors related to accessing a protected method, `config()`, within `bt_flatbuffer_helper.h`. 

```
/usr/local/include/behaviortree_cpp/flatbuffers/bt_flatbuffer_helper.h: In lambda function:
/usr/local/include/behaviortree_cpp/flatbuffers/bt_flatbuffer_helper.h:84:40: error: ‘BT::NodeConfig& BT::TreeNode::config()’ is protected within this context
   84 |     for (const auto& it : node->config().input_ports)
      |                                        ^
In file included from /usr/local/include/behaviortree_cpp/control_node.h:17,
                 from /usr/local/include/behaviortree_cpp/controls/parallel_node.h:17,
                 from /usr/local/include/behaviortree_cpp/behavior_tree.h:16,
                 from /usr/local/include/behaviortree_cpp/bt_factory.h:27,
/usr/local/include/behaviortree_cpp/tree_node.h:333:29: note: declared protected here
  333 |   [[nodiscard]] NodeConfig& config();
      |                             ^~~~~~
/usr/local/include/behaviortree_cpp/flatbuffers/bt_flatbuffer_helper.h:89:40: error: ‘BT::NodeConfig& BT::TreeNode::config()’ is protected within this context
   89 |     for (const auto& it : node->config().output_ports)
      |                                        ^
In file included from /usr/local/include/behaviortree_cpp/control_node.h:17,
                 from /usr/local/include/behaviortree_cpp/controls/parallel_node.h:17,
                 from /usr/local/include/behaviortree_cpp/behavior_tree.h:16,
                 from /usr/local/include/behaviortree_cpp/bt_factory.h:27,
/usr/local/include/behaviortree_cpp/tree_node.h:333:29: note: declared protected here
  333 |   [[nodiscard]] NodeConfig& config();
      |                             ^~~~~~
```

Attempting to access the config() function via a pointer to a object. Since the object pointed by node is non-const, the compiler will try to match the non-const version of the config() function first, which is protected, hence the error.

facontidavide, 2023-10-27:
Why is this compiling correctly in CI?

raviBhadeshiya, 2023-11-02:
Had a dive into the CI setup and noticed something that might have slipped through the cracks. It looks like the CI's scope is set to build bt_file_logger_v2 as per the settings in [CMakeLists.txt ](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/55149c86b85fb8f1c4685e856671399e41c23468/CMakeLists.txt#L119) Because of this, [bt_file_logger.cpp](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/55149c86b85fb8f1c4685e856671399e41c23468/src/loggers/bt_file_logger.cpp#L1) isn't getting compiled during the CI runs, which seems to be why we're missing the bt_flatbuffer_helper.h errors on the CI's radar.

Just to clarify, I'm not pointing fingers at the BehaviorTree.CPP library as a whole — it's a neat piece of work. It's just this little hiccup with the bt_flatbuffer_helper.h that's causing a bit of a snag.

closed 2024-01-17 by raviBhadeshiya