BehaviorTree/BehaviorTree.CPP #686: haltTree doesn't effectively halt tickWhileRunning

tony-p, 2023-10-30:
`haltTree` resets the tree status to `IDLE` which then in the `tickRoot` only breaks out of the inner loop, which is then restarted by the outer loop

```cpp
NodeStatus Tree::tickRoot(TickOption opt, std::chrono::milliseconds sleep_time)
{
  ...
  // Always run if IDLE
  while (status == NodeStatus::IDLE ||
         (opt == TickOption::WHILE_RUNNING && status == NodeStatus::RUNNING))
  {
   ...

    // Inner loop. The previous tick might have triggered the wake-up
    // in this case, unless TickOption::EXACTLY_ONCE, we tick again
    while( opt != TickOption::EXACTLY_ONCE &&
           status == NodeStatus::RUNNING &&
           wake_up_->waitFor(std::chrono::milliseconds(0)) )
    {
      status = rootNode()->executeTick();
      // Due to halt tree status is now IDLE, normal exiting operation would be SUCCESS or FAIL
    }
    ...
  }
  return status;
}

```

tony-p, 2023-10-30:
It may make more sense to make haltTree set the tree status to `FAILED` as it has not ended successfully

tony-p, 2023-10-31:
Yes a new `HALTED`/`HALTING` status would be more appropriate

facontidavide, 2023-10-31:
I see your point but... not gonna happen. A new status would mean rewriting ALL the control nodes and decorators

tony-p, 2023-10-31:
Yeah I understand is a pretty major API change, and that is why I originally suggested FAILED instead, but I also get your point that it is not each node that failed, (and can imagine could induce nasty side effects), so that status is only really relevant for the root node

closed 2026-09-20 by facontidavide