BehaviorTree/BehaviorTree.CPP #687: Question about disk write usage when using sqlite logger

OPerepadia, 2023-10-31:
I have enabled sqlite logger and trying to execute the following tree:

```xml
<root BTCPP_format="4">
    <BehaviorTree ID="MainTree">
      <ReactiveFallback>
        <AlwaysFailure/>
        <Sleep msec="100000"/>
      </ReactiveFallback>
    </BehaviorTree>
  </root>
```

When checking with `iotop`, I notice consistent disk write activity at around 1 MB/s rate. Is this a bug, or expected behavior?

facontidavide, 2023-11-01:
Because you are using a Reactive Node, **AlwaysFailure** is executed many times and that  is reported in the log.

I don't know what SQLite does under the hood, so I am not able to tell you if that is normal or not.

At which frequency is the tick invoked? Are you using `tickWhileRunning`?

OPerepadia, 2023-11-01:
I see. Maybe this is indeed how Sqlite works.

> At which frequency is the tick invoked? Are you using `tickWhileRunning`?

I'm actually calling `tickOnce()` at 20 Hz frequency.
I have just tried using `tickWhileRunning`, and disk usage increased to around 4.6 MB/s.

facontidavide, 2023-11-01:
ok. I don't think this should be an issue or that I  can do anything about it... so... closing the issue

closed 2023-11-01 by facontidavide