BehaviorTree/BehaviorTree.CPP #693: Allow custom implicit conversion of blackboard types

alsora, 2023-11-02:
Discussion on how to address the following problem:
Let's consider a behavior tree for a ROS 2 robotics application where several nodes require access to poses, positions, orientations, etc.
ROS 2 provides several data-types (messages) to handle those objects.
Let's say that we have the following BT nodes:
 - a node that gets the current robot pose
 - a node drives the robot towards a goal position
 - a node that rotates the robot to a certain orientation

Let's consider the following behavior tree (pseudo code)
```
  - sequence
  -  -  get_pose output_pose={pose}
  -  - drive_to_position input_position={pose}
  -  - rotate_to_orientation input_orientation={pose}
```

For this to work, you need that all the input and output ports are of the same type.
You could probably use either `geometry_msgs::msg::Pose` or `geometry_msgs::msg::PoseStamped`.

However, it may be confusing for a reader to see that a node that only cares about rotating the robot, requires a full pose as input.
This would also unnecessarily complicate the tree (let's say that we have a node that computes an orientation... now it needs to produce a full pose, with zeros for the translation part, in order to be able to feed it to the `rotate_to_orientation` node).

To address this problem, it would be nice to allow the definition of implicit conversions.
This way we could have something like :
 - the `get_pose` node would produce a `geometry_msgs::msg::PoseStamped` as output
 - the `drive_to_position` would declare an input port of type `geometry_msgs::msg::Position`. 
 - when checking the ports, we would note that although `geometry_msgs::msg::PoseStamped` and `geometry_msgs::msg::Position` are different types, the user provided a conversion function for them.

I think that this would improve readability and usability of complex behavior trees.

Any thoughts?

asasine, 2023-11-14:
Utilizing C++'s implicit or explicit conversions seems like a clean solution. The user would only need to define constructors on the target type that accept a single argument of the source type.

SteveMacenski, 2023-11-14:
Personally, I would not recommend that. That has alot of unintended consequences possible for subtle problems, particularly where frame transformations are concerned.

facontidavide, 2023-11-21:
doesn't sound as a good idea, and it is quite complex to implement.
Won't do for the time being

closed 2023-11-21 by facontidavide