BehaviorTree/BehaviorTree.CPP #706: Enable local variables in sub-tree

galou, 2023-11-30:
As sub-trees get more complicated, I tend to always use `_autoremap="true"` rather than writing a long list of remappings. Even if I would be OK with not using `_autoremap`, I cannot ensure that other users of the sub-tree I write also avoid to use it.

Now, I have a case where I compute some small trajectories identified with a shared prefix and then combine them. I do not know in advance how many of these trajectories will be in the blackboard but what I know for sure is that I don't want to share these small trajectories with other calls to the sub-tree.

I see two solutions:
- use a prefix to identify a local variable, e.g. `local{var}`
- add an attribute to the `BehaviorTree` element to list the variables that should not be shared even if `_autoremap` is used. Not so nice, IMHO.

I can implement something but I'd like to have some guidance on how to implement this.

facontidavide, 2023-12-02:
this seems like a very specific use case. I understand  that in your case could be annoying, but it is hard for me to find an elegant solution.

The prefix is doable, but seems too specific.

What about excluding from auto-remapping all the port names that start with "_",

"I_am_public"
"_I_am_private"

This feels very familiar for people developing C++

facontidavide, 2023-12-04:
What do you think?

galou, 2023-12-06:
This would be a useful addition that would help avoiding potential issues when using subtrees.

facontidavide, 2023-12-06:
done

galou, 2023-12-06:
Woah, that was quick! Thanks!

facontidavide, 2023-12-06:
That's what my wife said, excluding "thanks"