BehaviorTree/BehaviorTree.CPP #710: Enums evaluate to double not integral?

Roboterbastler, 2023-12-05:
I have come across the problem that I initialized a blackboard entry from an enum value (registered with `registerScriptingEnums`) but the entry seems to have real number type afterwards and a SwitchX node doesn't work properly because of this (string equality is not given, e.g. "0.0000" != "0").

I tried to trace this down a bit to this line and am wondering why it is casted to double and not an integral type, is there a reason?

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/ae95e680ab3c58f1e773f05d6372c10f41b546e8/include/behaviortree_cpp/scripting/operators.hpp#L75

facontidavide, 2023-12-06:
This is a known limitation of the scripting language, sorry. I will try to find a solution.
It would be nice if you can prepare failing unit test!!! :pray:

Roboterbastler, 2024-01-16:
Sorry for the late response, I proposed a failing unit test in #745.

facontidavide, 2024-01-16:
Thanks. I will try to address this in my next release cycle

facontidavide, 2024-01-23:
fixed. Thanks for the help. See unit test added for correct usage

closed 2024-01-23 by facontidavide