BehaviorTree/BehaviorTree.CPP #711: Real-time Visualization in Groot: is it free?

sgarciav, 2023-12-05:
Hello: It's not clear to me if the ability to monitor my nodes with Groot2 in real-time is part of the Basic package.

In the main [behaviortree.dev](https://www.behaviortree.dev/groot/) website, under the "Pricing" section, the Basic package states:
"Monitor and Log Visualizer limited to 20 Nodes"

However, in the [tutorial docs](https://www.behaviortree.dev/docs/tutorial-basics/tutorial_12_groot2/#adding-real-time-visualization-to-groot), there is a note that states:
"Currently, only the PRO version of Groot2 supports real-time visualization."

I am a user of the Basic package for Groot2. I have tried adding the line `BT::Groot2Publisher publisher(tree);` to my executor node, but it doesn't compile. So my questions are:

- Is the ability to monitor my nodes in real-time available for the Basic package?
- If yes, are there any other steps I need to take other than adding the `Groot2Publisher` line to my executor node?

Any information is more than welcomed. Thanks!

facontidavide, 2023-12-05:
It should work.can you tell me more about your error?

facontidavide, 2023-12-06:
I don't understand, everything looks ok.

If you use the same code from tutorial 12, does it work?

facontidavide, 2023-12-06:
Are you sure that `publisher` and `tree_` don't go out of scope? 
I am 90% positive that you are calling this inside a method and that `publisher` is destroyed immediately.

If that is the case, you may want to define an attribute in your class like

```cpp
std::unique_ptr<BT::Groot2Publisher> publisher_;
```

and change your code to:

```cpp
tree_ = factory_.createTreeFromFile(tree_file_path);
const unsigned port = 1667;
publisher_ = std::make_unique<BT::Groot2Publisher >(tree_, port);
```

closed 2023-12-06 by facontidavide

sgarciav, 2023-12-06:
This was exactly the problem. I tested it out and everything is working as expected. Thanks for the reply!

facontidavide, 2023-12-06:
[image]