BehaviorTree/BehaviorTree.CPP #716: Blackboard::set() failed

christophfroehlich, 2023-12-07:
Hello!

since a recent update, I get exceptions with a setup I used successfully in summer. I use the behavior tree action server from nav2, but I think the problem is within BT.CPP:

```
[BehaviorTreeEngine]: Behavior tree threw exception: Blackboard::set() failed: once declared, the type of a port shall not change. Declared type [double] != current type [bool]. Exiting with failure.
```
I didn't explicitly set blackboard variables in the c++ code. I think I narrowed it down to output ports, which I left empty in the xml. Is this possible? I mean one could say this is bad practice, but then a more useful exception would be great!

- ROS2 humble
- behaviortree_cpp_v3: 3.8.5
- navigation2: 1.1.12

facontidavide, 2023-12-13:
Branch 3.8 hasn't been changed in a while.

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/v3.8/CHANGELOG.rst

Unless you provide a specific example that helps me reproducing or debugging the issue, I can not help you

christophfroehlich, 2023-12-13:
Thanks for the answer, maybe this is related to the implementation of BT.CPP into nav2 and something changed there. I can try to create a minimal example without nav2 to see if I can reproduce this

facontidavide, 2023-12-19:
Closing. Please open again once you have a sample code / unit test I can run to reproduce the issue

closed 2023-12-19 by facontidavide