BehaviorTree/BehaviorTree.CPP #720: registerNodeType not accepting instance names

Lecrapouille, 2023-12-13:
Hi!  The [documentation](https://www.behaviortree.dev/docs/tutorial-basics/tutorial_01_first_tree) says:

```
Any instance of a TreeNode has a name. This identifier is meant to be human-readable and it doesn't need to be unique.
```

I have a project where I have to use two instances (`FreeRotor` and `FreeArm`) of the same action node `FreeCardAction` as depicted in the following screenshot:
[image]

The important point is that FreeCardAction is taking i.e. a boolean param in their constructor and the value shall be different for each instance.

The following code will produce a `Node not recognized` exception because I'm using the instance names instead of model name:

```
BT::BehaviorTreeFactory factory;
factory.registerNodeType<FreeCardAction>("FreeRotor", true);
factory.registerNodeType<FreeCardAction>("FreeArm", false);
```

- I find it sad to have to create several dummy classes inheriting from FreeCardAction for passing a different boolean value.
- registerNodeType should have been named registerModelName to follow Groot2 windows (else the logic would be to pass `Action` as template).
- We can interrogate ourselves on what is the utility of instance names:
  - the doc could be more explicit on how to use them.
  - add a registerInstanceName method or modifiy the behavior of registerNodeType to check first for a known instance names and if not found then check for a known model name.

facontidavide, 2023-12-13:
Please provide an example I can run or a unit test, and I will address this immediately

facontidavide, 2023-12-13:
This is really an error in the way you are using Groot2.
It looks like you are creating the model manually in Groot with the graphic editor instead of generating it with C++ as explained here: https://www.behaviortree.dev/docs/tutorial-basics/tutorial_12_groot2#the-treenodesmodel

Your code will create **two models**, NOT two instances of a single model. Therefore, the correct version would be:

```xml
<root BTCPP_format="4">
  <BehaviorTree ID="Untitled">
    <Sequence>
      <FreeRotor/>
      <FreeArm/>
    </Sequence>
  </BehaviorTree>

  <!-- Description of Node Models (used by Groot) -->
  <TreeNodesModel>
    <Action ID="FreeArm"/>
    <Action ID="FreeRotor"/>
  </TreeNodesModel>
</root>
```
[image]

facontidavide, 2023-12-13:
> I find it sad to have to create several dummy classes inheriting from FreeCardAction for passing a different boolean value.

Agree. You are doing it correctly

> registerNodeType should have been named registerModelName to follow Groot2 windows

I guess... 

> We can interrogate ourselves on what is the utility of instance names.

The instance name is just a label of a specific Node, that does not influence the execution. It is used for debugging purposes and logging only.

closed 2023-12-13 by facontidavide