BehaviorTree/BehaviorTree.CPP #725: [humble binary] BT: „missing port [value]” for blackboard “{goal}”

lmendyk, 2023-12-19:
I’m using Nav2 stack humble release which uses v.3.8 of BehaviorTree.cpp and I’m getting an error when using in a BT the following:
`<SetBlackboard value="{goal}" output_key="orignalGoal" />`
I get:

> Behavior tree threw exception: missing port [value]. Exiting with failure

But the {goal} is initialized and used by Nav2 and in my BT without any problem for example
`<ComputePathToPose goal="{goal}" path="{path}" planner_id="GridBased"/>`
So  I would expect that the value should not be missing in this example above.

In the Nav2 src I think this is the line which initializes the {goal}
```
...
bt_action_server_->loadBehaviorTree(bt_xml_filename))
...
// Update the goal pose on the blackboard
blackboard->set<geometry_msgs::msg::PoseStamped>(goal_blackboard_id_, goal->pose);

```

The bigger context is given in https://github.com/ros-planning/navigation2/issues/4016

facontidavide, 2023-12-19:
Just added a unit test, just in case, and everything works correctly.
If you think that there is a bug, please provide a failing test

lmendyk, 2023-12-19:
Thank you for prompt response.
I’m doing it as I hoped it could be a workaround for problem with port mapping when using subtree (I provided the ref to issue in Nav2 stack which provides bigger context  https://github.com/ros-planning/navigation2/issues/4016).

I’m just guessing that this is because the goal is not a String value but geometry_msgs::msg::PoseStamped so that may be the reason it does not work. I reported it initially in Nav2 as I suspected that this issue is between Nav2-BT. 

I got enthusiasm about BT as a tool for expressing complex logic using the concept of low-code/no-code so I may not be able to provide you the c++ test case as programing in c++ is sth which I wanted to avoid and use just XML defined BT.

But if SetBlackboard was not meant for copying a value from another key entry, than I’m stuck or need to learn how to build Nav2 with BT from sources to have   the bugs related to subtree ports remapping fixed (https://github.com/ros-planning/navigation2/issues/3640 )

Thank you anyway for quick response.

facontidavide, 2023-12-19:
I **believe** it should work with other types different from string

facontidavide, 2023-12-19:
I modified the unit test and you are right.
This functionality is not supported, but I think it should.

```c++
struct Point {
  double x;
  double y;
};

TEST(BlackboardTest, SetBlackboard_Issue725)
{
  BT::BehaviorTreeFactory factory;

  const std::string xml_text = R"(
  <root main_tree_to_execute = "MainTree"  >
    <BehaviorTree ID="MainTree">
      <SetBlackboard value="{first_value}" output_key="other_value" />
    </BehaviorTree>
  </root> )";

  factory.registerNodeType<DummyNodes::SaySomething>("SaySomething");
  factory.registerBehaviorTreeFromText(xml_text);
  auto tree = factory.createTree("MainTree");

  const Point point = {2,7};
  tree.blackboard_stack.front()->set("first_value", point);

  const auto status = tree.tickRoot();

  Point other_point = tree.blackboard_stack.front()->get<Point>("other_value");

  ASSERT_EQ(status, BT::NodeStatus::SUCCESS);
  ASSERT_EQ(other_point.x, point.x);
  ASSERT_EQ(other_point.y, point.y);
}
```

facontidavide, 2023-12-19:
Fixed in both version 3.8 and 4.4

closed 2023-12-19 by facontidavide