BehaviorTree/BehaviorTree.CPP #729: "Node not recognized" when running two trees from the same xml file

marcociara379, 2023-12-21:
Hello,
I am building a system with different ROS2 nodes. Each ROS2 node has a behavior tree that is used for the ROS2 node's internal logics.

Using Groot2, I generated a single xml file with different trees (each one with its own _tree ID_). The trees do not have the same condition nodes, i.e., a tree can have an exclusive condition node that another tree does not have.

I try to load the various trees via their _tree ID_ with:
```c++
factory.registerBehaviorTreeFromFile("./PATH/TO/TREE.xml");
auto tree = factory.createTree("TREE_ID");
```

However, I have `BT::RuntimeError` that specifies `Node not recognized: NAME_OF_THE_EXCLUSIVE_CONDITION_NODE`.

If I have different for each of the trees, this does not happen and everything goes fine.

Do you have any hint on how to solve the issue?

Thanks
Marco

facontidavide, 2024-01-10:
Give me a way to reproduce this and I will be happy to help

facontidavide, 2024-02-07:
Hi,

unless you provide an example code that I can reproduce, I am closing this

facontidavide, 2024-02-10:
Closing this for inactivity

closed 2024-02-10 by facontidavide