BehaviorTree/BehaviorTree.CPP #731: How to pass additional references?

GuraMumei, 2023-12-22:
I need to pass some additional references to the constructor of my class, but when I Call the function registerNodeType(),  it say:  "[registerNode]: since you have a static method providedPorts(), you MUST add a constructor with signature: (const std::string&, const NodeParameters&)"
My code:
```c++
class Test : public BT::SyncActionNode {
 public:
  BlackboardRead(const std::string& name, const BT::NodeConfiguration& config ,int& a) : BT::SyncActionNode(name, config) , a_(a){};

  static BT::PortsList providedPorts() {
     return {};
  }

  BT::NodeStatus  tick() override {
    ROS_INFO("a_: %d.", a_);
    return BT::NodeStatus::SUCCESS;
  };

  int& a_;
};
```

```c
int param= 2;
  factory.registerNodeType<rm_behaviortree::BlackboardRead>("BlackboardRead", param);
```

HenrikNilssonCode, 2024-01-09:
Checkout [Tutorial T08](https://www.behaviortree.dev/docs/tutorial-basics/tutorial_08_additional_args) ([Tutorial file](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/examples/t08_additional_node_args.cpp)). It looks like you are referencing an incorrect (old?) class, your constructor should have a signature with `BT::NodeConfig` NOT `BT::NodeConfiguration`. Also, in the provided code, it looks like the class is named Test, not BlackBoardRead, as you are trying to reference in the registration of your node type.

Thirdly (and probably what's actually causing your error), when passing a reference you must explicitly call `std::ref`. This is mentioned in the tutorial, but is not super obvious.

A correct snippet would look like this:
```c++
class BlackboardReadNode : public BT::SyncActionNode {
public:
  BlackboardReadNode(const std::string& name, const BT::NodeConfig& config, int& a)
    : BT::SyncActionNode(name, config) , a_(a) {};

  static BT::PortsList providedPorts() {
    return {};
  }

  BT::NodeStatus tick() override {
    ROS_INFO("a_: %d.", a_);
    return BT::NodeStatus::SUCCESS;
  };

  int& a_;
};
```

```c++
int param = 2;
factory.registerNodeType<rm_behaviortree::BlackboardReadNode>("BlackboardRead", std::ref(param));
```

facontidavide, 2024-01-09:
The answer given by @HenrikNilssonCode  is correct. If you want to avoid the `std::ref` pitfall, use a raw pointer or a shared pointer instead.

closed 2024-01-09 by facontidavide