BehaviorTree/BehaviorTree.CPP #734: ROS2 Humble, Updating to v4.X version results in conflict

Krzo99, 2023-12-29:
Hello,

So behaviortree_cpp_v3 comes with ros2 humble desktop installation. However, If I want to update to the newest version (for real time Groot2 visualization, future proofing,...) apt returns an error, 

Currently installed package: `ros-humble-behaviortree-cpp-v3 `
Command executed: `sudo apt install ros-humble-behaviortree-cpp`

Error: 
```
dpkg: error processing archive /var/cache/apt/archives/ros-humble-behaviortree-cpp_4.4.2-1jammy.20231205.182400_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/humble/bin/bt3_log_cat', which is also in package ros-humble-behaviortree-cpp-v3 3.8.5-1jammy.20231205.182408
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
```

Is this a bug? I would expect that it would be possible to have both versions installed at the same time, as they have different names. Is it possible to have both versions installed, for backwards compatibility of other ROS2 projects?

Thanks!

facontidavide, 2023-12-29:
Thanks for reporting.it is a bug.

I will fix it soon

closed 2024-01-10 by facontidavide

rebeccaRossRobotics, 2024-01-23:
Hi,

I've still getting this error
```
Unpacking ros-humble-behaviortree-cpp (4.4.3-2jammy.20231219.132918) ...
dpkg: error processing archive /var/cache/apt/archives/ros-humble-behaviortree-cpp_4.4.3-2jammy.20231219.132918_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/humble/bin/bt3_log_cat', which is also in package ros-humble-behaviortree-cpp-v3 3.8.6-1jammy.20231220.071041
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/ros-humble-behaviortree-cpp_4.4.3-2jammy.20231219.132918_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

Has the fix been released?

Thanks :)

Krzo99, 2024-01-23:
Hi @rebeccaRossRobotics 

As far as I know, the fix has been made (4.5.0), but only for ros2-iron. ros2-humble is still on 4.4.3:

[image]

I'm not sure when the humble version will be updated. Hopefully soon.

rebeccaRossRobotics, 2024-01-23:
@Krzo99 Ah ok thanks for informing!

facontidavide, 2024-01-23:
in version **4.5.x** that the ROS buildfarm hasn't released yet

Benjamin-Tan, 2024-01-25:
@Krzo99 it should be available after https://discourse.ros.org/t/preparing-for-humble-sync-and-patch-release-2024-01-26/35765/2

commit bdd41b2c (2024-01-10): fix #734