BehaviorTree/BehaviorTree.CPP #742: Re-ordering of ports and node TreeNodesModels in different Groot versions

schactus, 2024-01-13:
We have several developers using Groot2 to make/modify behavior trees. We do our best to keep everyone on the same version of Groot, but sometimes we get out of sync. It seems that the different versions of Groot have different methods of determining what the ordering of remapped entries in a call to a subtree, ports in a node, and TreeNodeModels should be. This causes MR's to be more challenging to review when changes are made using a version of Groot that doesn't match with the one previously used to create that tree file because there are many changes to the file that make no functional difference (things get reordered). 

We will continue to attempt to keep all developers on the same version of Groot, but it generally seems undesirable that things get re-ordered with different versions. Would it be possible to get some more information about how the ordering of those ports, models and remaps is determined in Groot? Should we anticipate that newer versions of Groot (after 1.5) continue to change the ordering of ports?

facontidavide, 2024-01-18:
yes, the order changed.
No, there is no plan to change it again

facontidavide, 2024-01-18:
Solution: don't use old version of Groot2.

closed 2024-01-18 by facontidavide