BehaviorTree/BehaviorTree.CPP #743: [4.x] How to Pause and resume the tree?

HBing110, 2024-01-15:
What is the best way to implement pause and resume the tree in the 4.x version of the program? Can you provide an example，thx

jdemir, 2024-01-16:
You are planning to pause/resume the tree from a higher level application, right? Why not creating a State Machine in higher level so you just tick the tree on Running state, while in Pause you will wait the user to click/request Start/Continue again.

HBing110, 2024-05-13:
> You are planning to pause/resume the tree from a higher level application, right? Why not creating a State Machine in higher level so you just tick the tree on Running state, while in Pause you will wait the user to click/request Start/Continue again.

Sorry, I still didn't understand what you meant.  Can you give a simple example？Thank you very much

jdemir, 2024-05-13:
so, the tree has the following method called [tickOnce](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/287eb59cefdc375cfd45bfa1431a840163e8199a/src/bt_factory.cpp#L604), let's the user to tick the tree only once when they want to. So, if you create an object that has the tree as a member variable and it has a run or update method so it can run synchronously (every X ms), you can double check within that run method whether anyone requested any Pause or Resume and tick the tree the accordingly. Roughly,

```
Update() {

if(userRequestedPause()){
    pause();
} else{
    tree.tickOnce();
}

}
```

HBing110, 2024-07-11:
> so, the tree has the following method called [tickOnce](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/287eb59cefdc375cfd45bfa1431a840163e8199a/src/bt_factory.cpp#L604), let's the user to [...]

Thank you for your patient answer. I understand now

closed 2024-07-11 by HBing110