BehaviorTree/BehaviorTree.CPP #748: Outdated static error message

Lecrapouille, 2024-01-17:
Hello !

I had an outdated message error in the following case (where by mistake I forget to add the `onHalted` method)

```
main.cpp:104:64:   required from here
/usr/local/include/behaviortree_cpp/bt_factory.h:363:19: error: static assertion failed: The given type can't be abstract
  363 |     static_assert(!std::is_abstract_v<T>, "The given type can't be abstract");
      |                   ^~~~~~~~~~~~~~~~~~~~~~
/usr/local/include/behaviortree_cpp/bt_factory.h:374:19: error: static assertion failed: [registerNode]: since you have a static method providedPorts(),
you MUST add a constructor with signature:
(const std::string&, const NodeParameters&)

  374 |     static_assert(!(has_static_ports_list && !param_constructable),
      |   
```

Here is the code:

```
class MoveToXYZAction : public BT::StatefulActionNode
{
public:

    MoveToXYZAction(const std::string& name, const BT::NodeConfig& config)
        : BT::StatefulActionNode(name, config)
    {
    }

    static BT::PortsList providedPorts()
    {
        return { BT::InputPort<Id>("arm") };
    }

    virtual BT::NodeStatus onStart() override
    {
        return BT::NodeStatus::RUNNING;
    }

    virtual BT::NodeStatus onRunning() override
    {
        return BT::NodeStatus::SUCCESS;
    }

    // To produce the previous error message
    // virtual void onHalted() override {}
};
```

Ok my class is still abstract because of missing override onHalted method but:
- `you MUST add a constructor with signature: (const std::string&, const NodeParameters&)` this message is outdated. It should be `NodeConfig` instead of NodeParameters.
- I have already the correct constructor so it's confusing to get the point between abstracted class, constructor and providedPorts(). The second message should not be displayed in all cases. I know this may be complex to separate error mesages.

facontidavide, 2024-01-17:
fixed.

closed 2024-01-17 by facontidavide

Lecrapouille, 2024-01-17:
@facontidavide Perfect ! I tried both cases: both messages are separated :)