BehaviorTree/BehaviorTree.CPP #752: Idea: compile-time if-then-else node

Lecrapouille, 2024-01-19:
Hi ! Sorry to disturb you again :)

What do you think about adding of a compiled-time if-then-else node ? This node will reuse the idea of the `Precondition` node with the `if-then-else` node:
- the `if-then-else` node is executed each `tick()` and depending on the result will execute the left or right child ?
- the `precondition` node, if I understand well, depending on the result of the scripted condition, will enable the child node or return failure/success. What I'm unsure of (I've not tested yet) if the precondition is called when the tree is build or at each tick(). I guess it's the option 2. And do I guess well, if I say the precondition code does not allow an "else" node branch ?

My idea of compiled-time if-then-else node is:

[image]

- the condition of the if-then-else node is interpreted ONLY when the tree is build. If the condition is true, then the left child (then-branch) is attached to the parent node of the if-then-else node, else it is the right branch (then-branch) which is attached (or failure/success if a single child is provided instead of two).
- it completes the precondition node by adding an "else-branch".
- the condition (`A && B`), likes for the precondition, will use the blackboard syntax (probably we can simplify be using a single boolean condition, so not necessary using script => so lighter version).
- The idea is to avoid having a tree with dead branches (never explored) if we know the condition is a constant (always true or always false) when we create the tree.
- Why this kind of tree ? Depending on the settings of the project, the tree may have different branches of behavior.
- We may reuse the idea of subtree by enabling it or disabling it.

For example, if the condition `A && B` is false (again: at compile time, meaning when the tree is build), the tree is build as:

[image]

facontidavide, 2024-01-19:
PR welcome to discuss this with actual code in front of us

Lecrapouille, 2024-01-22:
For the moment I'll keep using the if-then-else because, I dived in the code, but it's too deep for me and changes will inside part of the code that may impact (factory, just for one kind of node), so not sure if that is a good idea.

closed 2024-01-22 by Lecrapouille