BehaviorTree/BehaviorTree.CPP #757: Post script executes even when the node is skipped

ufrhaidar, 2024-01-30:
the **_post** Post Condition gets executed even when the node is skipped due to a Pre Condition.

From the documentation:
| Name | Description |
|-------------|---------|
| **_post**      | Execute this script, if the Node returned either SUCCESS or FAILURE |

I think it either needs to be fixed, or the documentation should say that the script would also execute when the node returns SKIPPED.

facontidavide, 2024-01-31:
this seems like a serious bug that I should address immediately.

Do you have a simple example / unit tests to reproduce the issue?

ufrhaidar, 2024-01-31:
[image]

This simple tree can reproduce the error.
It was printing good then bad

```
[behavior_tree_engine_node-1] [INFO] [1706740514.701689945] [behavior_tree_engine]: [UFR2_PrintString] Value: good
[behavior_tree_engine_node-1] [INFO] [1706740514.701755465] [behavior_tree_engine]: [UFR2_PrintString] Value: bad
```

ufrhaidar, 2024-01-31:
Found issue:

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/d4d6514966d416996efc624e567a45c0cc0bdeab/src/tree_node.cpp#L255

can be also fixed by checking if it is skipped before running this:

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/d4d6514966d416996efc624e567a45c0cc0bdeab/src/tree_node.cpp#L112

closed 2024-02-07 by facontidavide