BehaviorTree/BehaviorTree.CPP #758: change signature of registerNodeType to accept StringView

stewpend0us, 2024-01-31:
As far as I can tell...right now ```registerNodeType``` requires a ```const std::string& ID``` but ultimately just makes a copy of the string anyway. Why not use StringView instead?

That way I can keep a ```static constexpr StringView ID``` on my class and pass it directly into ```registerNodeType<MyType>(MyType::ID, ...)``` without having to jump through the (very small) extra hoop of defining a ```static const std::string```

Better yet (maybe?) allow me to optionally define my class with a ```static constexpr StringView ID``` property that _just_ gets picked up by the register function?

facontidavide, 2024-01-31:
feel free to submit a PR

facontidavide, 2024-02-01:
So, I see that there **could** be 1 copy of a string less during initialization, but since this run-time optimization only happens at the very beginning, when the node is registered, it is not a good reason to apply a non-ABI compatible change

closed 2024-02-01 by facontidavide

stewpend0us, 2024-02-01:
Sorry. I can do what I need with the "registrationName()" method.