BehaviorTree/BehaviorTree.CPP #767: `logic_error` on `BT::BehaviorTreeFactory::registerNodeType()` When Setting `InputPort` `default_value` to `nullptr` After Commit 789ce6ea0ad3627923bd2389b8fb9199ffab6d84

KentaKato, 2024-02-13:
Following the changes made in commit 789ce6ea0ad3627923bd2389b8fb9199ffab6d84, setting the `default_value` of an `InputPort` to `nullptr` now results in a `logic_error` during the execution of `BT::BehaviorTreeFactory::registerNodeType()`.

The program terminates with the following error message:

```bash
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Aborted (core dumped)
```

This issue indicates a problem with handling `nullptr` as the `default_value` for `InputPort` in the context of `BT::BehaviorTreeFactory::registerNodeType()`, which appears to be a new behavior following the specified commit.

facontidavide, 2024-02-13:
Please provide a unit test. Also what is the type of the port?
What is your intent with a nullptr?

KentaKato, 2024-02-13:
@facontidavide 
Thank you for your prompt reply.

I cannot provide a unit test right away, but I will first address the latter part of your question.

I had set `nullptr` as the default value for `shared_ptr` as shown below. `T` is a class with a high instantiation cost, and an instance is only created within the Node if it's `nullptr`. Is this usage unexpected?

```cpp
    static BT::PortsList providedPorts()
    {
        return {
            BT::InputPort<std::shared_ptr<T>>("test", nullptr, "description")
    };
```

facontidavide, 2024-02-13:
Check 83fce13a38efe81f2dd5713328f94942f32f8c9a

If the problem persists, please tell me how to extend the unit test **Default_Issues_767_768** to cover your case.

closed 2024-02-13 by facontidavide

KentaKato, 2024-02-14:
@facontidavide 
Thank you for addressing this so quickly. I've verified that the problem has been solved.