BehaviorTree/BehaviorTree.CPP #782: Pixi Windows failing in CI

facontidavide, 2024-03-08:
@tony-p , any suggestion?

I am not sure what the error is

tony-p, 2024-03-08:
From looking at the error message my best guess would be something has changed in the path or how to specify the temporary environment variable.

If I remember correctly this was needed on windows as the test executable and library were in different folders compared to the linux build.

When i get a chance I can try take a look if still stuck.

if you aren’t against it, it would probably be more robust to create a pixi.toml file in the repo so it can be used as pixi project and then CI wouldn’t have to try set it all up on the fly.

it would just live alongside the ros and conan files giving another way to interact with the project

facontidavide, 2024-03-13:
go ahead with any modification that you consider appropriate

tony-p, 2024-03-20:
I have made a start here https://github.com/BehaviorTree/BehaviorTree.CPP/pull/791

This fixes the issue you encountered, but I see it now has linux build issues. I get these issues with my ROS builds and is to do with the sqlite and zeromq not being added to the rpath properly.

Locally on windows I get Json warnings/errors when building and the json tests fail

tony-p, 2024-03-26:
@facontidavide The PR above fixes all pixi build related issues. However both the windows and pixi windows builds fail for the new json conversion so have created a new issue https://github.com/BehaviorTree/BehaviorTree.CPP/issues/793

closed 2024-04-02 by facontidavide