BehaviorTree/BehaviorTree.CPP #786: complex objects input/output ports, always need to provide a converter?

PaddyCube, 2024-03-15:
Hi there,
I'm developing a ros2 robot with some floor coverage. My ros2 eco system provides a set of services:
- srv1: read the area to cover (array of outline points and keepout zones)
- srv2: build a coverage path based on array of points
- nav2 action to follow the previously created path

To my understanding srv1 and srv2 can be implemented as RosServiceNode. srv2 will need a input port to get the area to cover. It will also need an output port to write the created paths (array of nav_msgs::Path) back. 

I'm struggling with how to import and export complex data. 

```
#ifndef _GET_COVERAGE_PATH_HPP
#define _GET_COVERAGE_PATH_HPP

#include <behaviortree_ros2/bt_service_node.hpp>
#include "slic3r_coverage_planner/srv/plan_path.hpp"
#include "slic3r_coverage_planner/msg/path.hpp"
#include "ros2mower_msgs/msg/map_area.hpp"

using namespace BT;

class GetCoveragePath : public RosServiceNode<slic3r_coverage_planner::srv::PlanPath>
{
public:
  GetCoveragePath(const std::string &name,
                  const NodeConfig &conf,
                  const RosNodeParams &params);

  static BT::PortsList providedPorts()
  {
    // compiles w/o error
    //return providedBasicPorts({BT::InputPort<std::string>("mowArea"),
    //                           BT::OutputPort<std::string>("path")});
    return providedBasicPorts({BT::InputPort<ros2mower_msg::msg::MapArea>("mowArea"),
                               BT::OutputPort<nav_msgs::msg::Path>("path")});
  }

  bool setRequest(Request::SharedPtr &request) override;
  NodeStatus onResponseReceived(const Response::SharedPtr &response) override;
};

#endif
```

If I define them as std::string, my code compiles without any error. 

```
/home/ros/ros2_ws/install/behaviortree_cpp/include/behaviortree_cpp/basic_types.h:436:41: note: candidate: ‘template<class T, class DefaultT> std::pair<std::__cxx11::basic_string<char>, BT::PortInfo> BT::InputPort(BT::StringView, const DefaultT&, BT::StringView)’
  436 | inline std::pair<std::string, PortInfo> InputPort(StringView name,
      |                                         ^~~~~~~~~
/home/ros/ros2_ws/install/behaviortree_cpp/include/behaviortree_cpp/basic_types.h:436:41: note:   template argument deduction/substitution failed:
In file included from /home/ros/ros2_ws/src/ros2mower/ros2mower_behavior/src/main.cpp:7:
/home/ros/ros2_ws/src/ros2mower/ros2mower_behavior/include/behaviors/GetCoveragePath.hpp:20:74: error: template argument 1 is invalid
   20 |     return providedBasicPorts({BT::InputPort<ros2mower_msg::msg::MapArea>("mowArea"),
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
/home/ros/ros2_ws/src/ros2mower/ros2mower_behavior/include/behaviors/GetCoveragePath.hpp:20:30: error: cannot convert ‘<brace-enclosed initializer list>’ to ‘BT::PortsList’ {aka ‘std::unordered_map<std::__cxx11::basic_string<char>, BT::PortInfo>’}
   20 |     return providedBasicPorts({BT::InputPort<ros2mower_msg::msg::MapArea>("mowArea"),
      |            ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   21 |                                BT::OutputPort<nav_msgs::msg::Path>("path")});
      |                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ros/ros2_ws/src/ros2mower/ros2mower_behavior/include/behaviors/SetMission.hpp:4,
                 from /home/ros/ros2_ws/src/ros2mower/ros2mower_behavior/src/main.cpp:5:
/home/ros/ros2_ws/install/behaviortree_ros2/include/behaviortree_ros2/bt_service_node.hpp:94:49: note:   initializing argument 1 of ‘static BT::PortsList BT::RosServiceNode<ServiceT>::providedBasicPorts(BT::PortsList) [with ServiceT = slic3r_coverage_planner::srv::PlanPath; BT::PortsList = std::unordered_map<std::__cxx11::basic_string<char>, BT::PortInfo>]’
   94 |   static PortsList providedBasicPorts(PortsList addition)
      |                                       ~~~~~~~~~~^~~~~~~~
gmake[2]: *** [CMakeFiles/ros2mower_behavior.dir/build.make:160: CMakeFiles/ros2mower_behavior.dir/src/behaviors/GetCoveragePath.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[2]: *** [CMakeFiles/ros2mower_behavior.dir/build.make:76: CMakeFiles/ros2mower_behavior.dir/src/main.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/ros2mower_behavior.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```

According to tutorials, it seems that I need to provide a converter to serialize the message into a string. This can be a huge task and I wonder if I can overcome this with [zero copy access?](https://www.behaviortree.dev/docs/tutorial-advanced/tutorial_13_blackboard_reference)

Maybe it is a bad practice to create a BT RosServiceNode for each and every service I need? Would it be ok to create a RosActionNode instead, which takes care of all necessary service calls? I need to call multiple services in sequence and it doesn't feel efficient to build a converter for each of them

This is how service definition looks like for srv2

```
uint8 FILL_LINEAR=0
uint8 FILL_CONCENTRIC=1
uint8 fill_type
float64 angle
float64 distance
float64 outer_offset
uint8 outline_count
uint8 outline_overlap_count
geometry_msgs/Polygon outline
geometry_msgs/Polygon[] holes
---
Path[] paths
```

facontidavide, 2024-03-15:
No, they should not need to implement **convertFromString**

**I dont know why your code is not compiling, it looks correct to me**

> Maybe it is a bad practice to create a BT RosServiceNode for each and every service I need?

It depends... RosServiceNode will save a lot of boiler plate. If you want to to create your own. the amount of CPU and bandwidth you use will be the same, but you will be responsible for implementing it

If you can provide a minimal example that reproduce the issue I can help, otherwise there is nothing I can do

PaddyCube, 2024-03-15:
Arrrgg, unbelievable. There was a typo. The message namespace of InputPort is ros2mower_msgs instead ros2mower_msg

    return providedBasicPorts({BT::InputPort<ros2mower_msgs::msg::MapArea>("mowArea"),
                               BT::OutputPort<nav_msgs::msg::Path>("path")})

But thanks for the hint about RosServiceNode. I'll try to continue to create each service as RosServiceNode instead of implement everything, inclusive all necessary service calls as a RosActionNode

closed 2024-03-15 by PaddyCube