BehaviorTree/BehaviorTree.CPP #810: [3.8] ReactiveSequence not halting children after a running child anymore

afrixs, 2024-04-23:
There was recently a backporting update of 3.8 where [this line](https://github.com/BehaviorTree/BehaviorTree.CPP/blame/95210b01f12f30fc25946c03db0fed0b826be1fd/src/controls/reactive_sequence.cpp#L31) changed from
```
for (size_t i = index + 1; i < childrenCount(); i++)
```
to
```
for (size_t i = 0; i < index; i++)
```
making ReactiveSequence unable to halt children that were previously running after a child whose state has just switched from SUCCESS to RUNNING. Meanwhile the bug seems to have been [fixed](https://github.com/BehaviorTree/BehaviorTree.CPP/blame/1bacc3f459ffcfa9d1a08bfd69994422f773ba5b/src/controls/reactive_sequence.cpp#L47) in 4.x, but the fix has not been backported to 3.8. Is it possible to backport the fix of this problem as well? Thank you

facontidavide, 2024-05-08:
send a PR please

afrixs, 2024-05-10:
I don't dare to make an overall backport merge PR, but #817 is a backport of the particular commit that fixes the problem (+ a small optional defaults-changing commit that keeps the backward compatibility to v3.8.5)

commit db25cfbd (2024-05-13): Fixed #810 - halting of subsequent nodes in ReactiveSequence/Fallback (#817)

* ReactiveSequence and ReactiveFallback will behave more similarly to 3.8

* Reactive Sequence/Fallback defaulting to allow multiple async nodes

---------
closed 2024-05-14 by afrixs