BehaviorTree/BehaviorTree.CPP #812: InputPort and OutputPort not connected.

PeterQFR, 2024-04-26:
Hi I have a condition node that writes a string to an outputPort based on some condition (an input from ROS2). and I have other condition nodes that check the value of that port via inputPort with the same key. this is used to switch different parts of the BT on or off.  

However, when I check for a good return from the inputPort and outputPort via 
```
std::string value;
if (inputPort("key", value))
{
    //Handle success 
}
else
{
   //handle failure.
}
```
it always returns false, and value is not updated. 

In both nodes I am  returning the BT::PortsList via providedPorts method. 

Is there any other initialisation I have missed out? is there a debug method to print the blackboard?

facontidavide, 2024-04-26:
This functionality has been tested, so I guess that it is a problem of either:

1. Writing and reading the port happens in different subtrees and you are not doing the remapping
2. XML not using the curly braces {} to specify that is an entry in the blackboard.

If you can share a specific code that I can use to reproduce the issue, I can be more specific, otherwise I can't.

Please either provide this code example or close the issue :smiling_face_with_tear:

PeterQFR, 2024-04-26:
Thanks, I found that it was essential to specify the port in the xml. It wasn't initially clear that this was an essential requirement, as I don't use inputs via xml in this use case. Using the curly braces syntax to specify the port to itself eg
```
<Condition1 next_action="{next_action}"/>
```
was required for the next_action port to be setup correctly on the node using the OutputPort  and the receiver node using the InputPort. 

Perhaps making it clearer that specifying the port in xml is essential in the documentation, or throw a big fat warning/error. It wasn't obvious even in the source that this was a requirement, and it fails silently.

I'll save my tears for now  :-)

facontidavide, 2024-04-28:
If you submit a concrete example where the error happens silently, I will be happy to review it and see if I can/should do better.

Idon't believe it is failing "silently", since **getInput** fails (should) with an error message

closed 2024-04-28 by facontidavide