BehaviorTree/BehaviorTree.CPP #814: Compiler warning - useless cast

jwdinius, 2024-04-30:
Hello!

I am observing the following compilation error when including `behaviortree_cpp/bt_factory.h` in my source file.

```bash
--- stderr: mission_manager                                    
In file included from /opt/ros/humble/include/behaviortree_cpp/utils/safe_any.hpp:23,
                 from /opt/ros/humble/include/behaviortree_cpp/basic_types.h:14,
                 from /opt/ros/humble/include/behaviortree_cpp/tree_node.h:21,
                 from /opt/ros/humble/include/behaviortree_cpp/control_node.h:17,
                 from /opt/ros/humble/include/behaviortree_cpp/controls/parallel_node.h:17,
                 from /opt/ros/humble/include/behaviortree_cpp/behavior_tree.h:16,
                 from /opt/ros/humble/include/behaviortree_cpp/bt_factory.h:28,
                 from /ws/src/mission_manager/include/mission_manager/mission_manager.hpp:7,
                 from /ws/src/mission_manager/src/mission_manager.cpp:5:
/opt/ros/humble/include/behaviortree_cpp/contrib/any.hpp: In member function ‘linb::any& linb::any::operator=(linb::any&&)’:
/opt/ros/humble/include/behaviortree_cpp/contrib/any.hpp:94:9: warning: useless cast to type ‘class linb::any’ [-Wuseless-cast]
   94 |         any(std::move(rhs)).swap(*this);
      |         ^~~~~~~~~~~~~~~~~~~
```

The offending method is [here](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/include/behaviortree_cpp/contrib/any.hpp#L94).

The argument type is `any&&`, so the C-style cast is unnecessary and confusing IMO.

facontidavide, 2024-04-30:
This is a 3rd party library: https://github.com/thelink2012/any

I will not modify it myself. If that warning is fixed there, I will update my version

Ryanf55, 2024-04-30:
If `std::any` is now part of the C++ standard as of C++17, why do we need the external library?

facontidavide, 2024-04-30:
because that implementation has a small object optimization of 16 bytes.

I can noit be sure about how **std::any** is inmplemented by a compiler

jwdinius, 2024-04-30:
FYI - https://github.com/thelink2012/any/pull/25

jwdinius, 2024-05-05:
> FYI - [thelink2012/any#25](https://github.com/thelink2012/any/pull/25)

The PR was merged upstream.

facontidavide, 2024-05-08:
fixed

closed 2024-05-08 by facontidavide