BehaviorTree/BehaviorTree.CPP #815: t19 executing problem

siyiovo, 2024-05-04:
I just git the latest version and paste t19_global_blackboard.cpp to my own workspace,but it crashed when I try to run it.
The issue is as follow:
terminate called after throwing an instance of 'nonstd::expected_lite::bad_expected_access<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >'
  what():  bad_expected_access

it seems that BT enter an endless loop.

siyiovo, 2024-05-04:
I just find the same [issue](https://github.com/BehaviorTree/BehaviorTree.CPP/issues/199) on May, 2020, in the previous version needs port remapping, but in the `t19_globak_balckcoard.cpp` which mention that `It is possible (since version 4.6) to create a global BB, accessible from everywhere without remapping.`I am sure that I compile v4.6.0 correctly.

facontidavide, 2024-05-04:
That issue was closed because no one provided a way to reproduce it.

Please give me a way to reproduce the issue

facontidavide, 2024-05-04:
Send me the code that crashed or suggest a failing unit test. It is IMPOSSIBLE for me to find the issue otherwise

siyiovo, 2024-05-04:
> Send me the code that crashed or suggest a failing unit test. It is IMPOSSIBLE for me to find the issue otherwise

closed 2024-05-04 by siyiovo

facontidavide, 2024-05-04:
Your reply is a quote of my text and it doesn't add anything

siyiovo, 2024-05-04:
> Your reply is a quote of my text and it doesn't add anything

I guess I find an exact way to run perfectly...I just compile again😂🤣