BehaviorTree/BehaviorTree.CPP #816: Segmentation fault on BT destruction.

Krzo99, 2024-05-08:
I came across a problem that only happens around 5-10% of the time.

It appers that a segmentation fault is thrown on destruction of BT::Tree object.

My setup is as follows:

First, I spin up a seperate thread, that will run Behavior Trees:  `std::thread{ std::bind(&AutonomousRobotControl::StartbehaviorTreeThread, this) }.detach();`

In the thread, I construct all objects:
```
BT::BehaviorTreeFactory factory;

factory.registerNodeType<BT_Node::Navigate>("Navigate", navigator_);
factory.registerNodeType<BT_Node::UpdateGoal>("UpdateGoal", navigator_);
....
factory.registerSimpleCondition("isLocalized", std::bind(BT_Condition::isLocalized, rtabmap_manager_));
factory.registerSimpleCondition("isColdStart", std::bind(BT_Condition::isColdStart, rtabmap_manager_));
....

auto tree = factory.createTreeFromFile(behavior_tree_xml_);
BT::Blackboard::Ptr blackboard = tree.rootBlackboard();

blackboard->set("current_action", RobotState::IDLE);
....

std::unique_ptr<BT::StdCoutLogger> logger;
if (debug_mode_)
    logger = std::make_unique<BT::StdCoutLogger>(tree);

 BT::Groot2Publisher groot2_telemetry_publisher(
    tree, 
    1667 + parent_->get_parameter("robot_id").as_int() * 2);
```

Then, I start the behavior loop:

```
while(rclcpp::ok()) {
            if (is_stopping_) {
                is_running_ = false;
                break;
            }
            tree.tickOnce();

        rate.sleep();
    }

```

Behavior tree execution is stopped, once is_stopping_ is set from the main thread. 
(There is a mutex to synchronize access, I just didn't include it in this code snip)

The errors I'm getting is always a bit different every time, however what seems to be common denominator, in that they are all based around (and they also happen) when behavior tree destructs, as seen in the logs.

I'm not sure what the problem is. Would it be possible that I'm stopping the behavior tree execution wrongly?
Thanks for your help!

Down bellow, I provide 2 Callgrind traces, and one gdb.

Callgrind 1:
```
[valgrind-7] ==139311== Process terminating with default action of signal 11 (SIGSEGV)
[valgrind-7] ==139311==  Access not within mapped region at address 0x1C00002D8
[valgrind-7] ==139311==    at 0x5376857: _int_free (malloc.c:4590)
[valgrind-7] ==139311==    by 0x5379452: free (malloc.c:3391)
[valgrind-7] ==139311==    by 0x563AD46: std::_Hashtable<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::TreeNodeManifest>, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::TreeNodeManifest> >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >::clear() (new_allocator.h:145)
[valgrind-7] ==139311==    by 0x5625E37: BT::BehaviorTreeFactory::~BehaviorTreeFactory() (hashtable.h:1532)
[valgrind-7] ==139311==    by 0x4875B14: terrascout_control::AutonomousRobotControl::StartbehaviorTreeThread() (autonomous_robot_control.cpp:237)
[valgrind-7] ==139311==    by 0x5164252: ??? (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30)
[valgrind-7] ==139311==    by 0x5368AC2: start_thread (pthread_create.c:442)
[... 121 lines cut ...]
[gdb-7] #31 std::_Destroy<std::shared_ptr<BT::Tree::Subtree> >(std::shared_ptr<BT::Tree::Subtree>*) (__pointer=0x7fff1803a070)
[gdb-7]     at /usr/include/c++/11/bits/stl_construct.h:149
[gdb-7] #32 std::_Destroy_aux<false>::__destroy<std::shared_ptr<BT::Tree::Subtree>*>(std::shared_ptr<BT::Tree::Subtree>*, std::shared_ptr<BT::Tree::Subtree>*)
[gdb-7]     (__last=0x7fff1803a090, __first=0x7fff1803a070)
[gdb-7]     at /usr/include/c++/11/bits/stl_construct.h:163
[gdb-7] #33 std::_Destroy<std::shared_ptr<BT::Tree::Subtree>*>(std::shared_ptr<BT::Tree::Subtree>*, std::shared_ptr<BT::Tree::Subtree>*)
[gdb-7]     (__last=0x7fff1803a090, __first=<optimized out>)
[gdb-7]     at /usr/include/c++/11/bits/stl_construct.h:196
[gdb-7] #34 std::_Destroy<std::shared_ptr<BT::Tree::Subtree>*, std::shared_ptr<BT::Tree::Subtree> >(std::shared_ptr<BT::Tree::Subtree>*, std::shared_ptr<BT::Tree::Subtree>*, std::allocator<std::shared_ptr<BT::Tree::Subtree> >&)
[gdb-7]     (__last=0x7fff1803a090, __first=<optimized out>)
[gdb-7]     at /usr/include/c++/11/bits/alloc_traits.h:848
[gdb-7] #35 std::vector<std::shared_ptr<BT::Tree::Subtree>, std::allocator<std::shared_ptr<BT::Tree::Subtree> > >::~vector()
[gdb-7]     (this=0x7fff137c3660, this=<optimized out>)
[gdb-7]     at /usr/include/c++/11/bits/stl_vector.h:680
[gdb-7] #36 BT::Tree::~Tree() (this=0x7fff137c3660, this=<optimized out>)
[gdb-7]     at ./src/bt_factory.cpp:573
[gdb-7] #37 0x00007ffff7f82b0b in terrascout_control::AutonomousRobotControl::StartbehaviorTreeThread() (this=0x555556850750)
[gdb-7]     at /home/domen/Documents/TerraScout/core/src/terrascout/terrascout_control/src/autonomous_robot_control.cpp:237
[gdb-7] #38 0x00007ffff74dc253 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
[gdb-7] #39 0x00007ffff7094ac3 in start_thread (arg=<optimized out>)
[gdb-7]     at ./nptl/pthread_create.c:442
[gdb-7] #40 0x00007ffff7126850 in clone3 ()
[gdb-7]     at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81

```

facontidavide, 2024-05-08:
Did you upgrade recently to version 4.6?

If that is the case, it might be a problem of ABI compatibility. Try recompiling from scratch everything, including the plugins.

Otherwise, we can try using **address sanitizer** to get more information.

The info you shared is not giving me any hypothesis, for the time being.

Krzo99, 2024-05-08:
Ok, so it appears that the problem was the fact that I was manually .reset() a shared_ptr with a custom class that was used in some Nodes in the tree, at the end of thread() function. 
I'm guessing there was a possibility for a race condition here. I just moved the .reset() call to after the thread joined.
I also changed a detachable thread to a joinable one, as that feels more accurate.

Currently, after 2h of testing, there hasn't been another error like this, so I'm guessing I solved it :)