BehaviorTree/BehaviorTree.CPP #818: Compile error in flatbuffers/stl_emulation with new GCC version

ltoenning, 2024-05-16:
**Describe the bug:**
When compiling ``master`` with GCC 14.1.1, I am getting the following error message:

```
[...]/BehaviorTree.CPP/include/behaviortree_cpp/flatbuffers/stl_emulation.h:549:12: error: assignment of read-only member ‘flatbuffers::span<T, Extent>::count_’
  549 |     count_ = other.count_;
      |     ~~~~~~~^~~~~~~~~~~~
```

This was fixed in flatbuffers with https://github.com/google/flatbuffers/pull/7227. It seems that the new GCC version somehow detects this error now as well.

This also affects the v3.8 branch.

**How to Reproduce:**

Compile BehaviorTree.CPP with GCC 14.1.1

facontidavide, 2024-05-20:
I am going toi remove entirely flatbuffers from the public API.

This means deprecating the old FileLogger in favor of FileLogger2, that is much simpler and supported by Groot2.

In terms of version 3.8, I will not mainatin it anymore. If you want to submit a PR for that, I will be happy to review it.

facontidavide, 2024-05-20:
Better to reduce the amount of changes, since I don't want to support it