BehaviorTree/BehaviorTree.CPP #822: Use rvalues with the Input/Ouput ports.

captain-yoshi, 2024-05-18:
**Is your feature request related to a problem? Please describe.**
I use a library that relies on `unique pointers`. My goal is to use unique pointers with `BT Ports` input + output. This does not work because std::any (which should be renames std::almost_any) accepts only CopyConstructible types.

**Describe the solution you'd like**
I would like a way to pass `std::unique_ptr<T>` in BT ports (or if feasible any rvalue reference).

```cpp
// Maybe by adding these new ports, we could use another mechanism to std::any ...
BT::InputRValuePort
BT::OutputRValuePort

// Alternative
BT::InputUniquePtrPort
BT::OutputUniquePtrPort
```
Obviously an `OutputRValuePort` could only be added to the `InputRValuePort`. It would be nice to view this in Groot2, the user would know that you can't reuse this variable when adding to the `InputRValuePort`.

**Describe alternatives you've considered**
I currently pass a shared_ptr (constructed from the unique_ptr) to the OutputPort. When the pointer is retrieved in a BT Node, i.e. with the `getLockedPortContent`, I set the any_ptr to nullptr. This is to ensure that subsequent BT Nodes does not have access to this unique_ptr in disguise. Finally, a unique_ptr is recreated from the shared_ptr.

**Additional context**
Maybe you have other suggestions to tackle this problem.

facontidavide, 2024-05-20:
> I would like a way to pass std::unique_ptr<T> in BT ports

The blackboard uses value semantic.

If you think about it, that is not compatible with **std::unique_ptr**,  because, when reaging from the port, you woult need to take ownership of the object.
Of course you may use only reference semantic to set/get the port, but that goes against the current API.

Even if your library returns a `unique_ptr`,  as you already mentioned that can be converted into a `share_ptr`. That seems to me the easiest solution and I don't see any problem with it. BUT:

> Finally, a unique_ptr is recreated from the shared_ptr.

Why????

facontidavide, 2024-05-20:
I am currently marking this as "wontfix" and close. This seems like a large and complex change and the use case is not compellingenough to motivate such a big change.

PR are welcome

closed 2024-05-20 by facontidavide

captain-yoshi, 2024-05-20:
> Finally, a unique_ptr is recreated from the shared_ptr.
> 
> Why????

I need to move it into another function that takes a unique pointer as an argument.

Maybe I can encapsulate some of the low-level logic. Will try different things and maybe submit a PR.

Thanks for taking a look at my issue.