BehaviorTree/BehaviorTree.CPP #823: Impossible to write to global blackboard using output ports

robin-mueller, 2024-05-19:
In my application I am trying to write to a global blackboard entry using `setOutput` of a tree node, but an entry in the local blackboard is created instead.

Let an example be:

```xml
<BehaviorTree ID="BugExample">
    <Sequence>
        <Script code="@global_var:=0"/>
        <SetState in="1"/>
        <GetState out="{@global_var}"/>
    </Sequence>
</BehaviorTree>
```

The nodes `SetState` and `GetState` set respectively get the value of an internal state variable using the corresponding ports.

I'd suggest to implement an example like this in the associated [test](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/287eb59cefdc375cfd45bfa1431a840163e8199a/tests/gtest_blackboard.cpp#L566-L608) to verify this issue.

robin-mueller, 2024-05-19:
I've added the test in the pull request above.

I currently don't have too much time to look into it, but I assume the issue to have its origin [here](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/287eb59cefdc375cfd45bfa1431a840163e8199a/include/behaviortree_cpp/blackboard.h#L203). `storage_` is searched for a key `@global_var` which obviously cannot be found.

facontidavide, 2024-05-20:
fixed

closed 2024-05-20 by facontidavide