BehaviorTree/BehaviorTree.CPP #844: BT::Blackboard::set<>() crash from pthread_mutex_lock

yanfushi, 2024-07-03:
Hello,
I want to set an nav_msgs::msg::path into blackboard in a control node (similar with pipeline_sequence)

  while (current_child_idx_ == 0) {
    auto status = children_nodes_[current_child_idx_]->executeTick();
    switch (status) {
      case BT::NodeStatus::FAILURE:
      case BT::NodeStatus::SUCCESS: {
        ControlNode::haltChildren();
        nav_msgs::msg::Path empty_path;
        config().blackboard->set<nav_msgs::msg::Path>("lpath", empty_path);
        last_child_idx_ = current_child_idx_;
        current_child_idx_ = 1;
        return BT::NodeStatus::RUNNING;
        break;
      }
...
and got a backtrace like this
[planner_and_controller_server-6] E0703 15:18:28.844405 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe952509ee0 BT::Blackboard::set<>()
[planner_and_controller_server-6] E0703 15:18:28.844990 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe8f41713ac nav2_behavior_tree::CollectTaskSequence::tick()
[planner_and_controller_server-6] E0703 15:18:28.848073 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe950d2ff2a BT::TreeNode::executeTick()
[planner_and_controller_server-6] E0703 15:18:28.850723 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe950d4f3a8 BT::ReactiveFallback::tick()
[planner_and_controller_server-6] E0703 15:18:28.853137 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe950d2ff2a BT::TreeNode::executeTick()
[planner_and_controller_server-6] E0703 15:18:28.856010 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe950d4fa86 BT::SequenceNode::tick()
[planner_and_controller_server-6] E0703 15:18:28.858567 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe950d2ff2a BT::TreeNode::executeTick()
[planner_and_controller_server-6] E0703 15:18:28.860822 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe950d4f3a8 BT::ReactiveFallback::tick()
[planner_and_controller_server-6] E0703 15:18:28.863410 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe950d2ff2a BT::TreeNode::executeTick()
[planner_and_controller_server-6] E0703 15:18:28.866350 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe950d4ee3a BT::ReactiveSequence::tick()
[planner_and_controller_server-6] E0703 15:18:28.868615 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe950d2ff2a BT::TreeNode::executeTick()
[planner_and_controller_server-6] E0703 15:18:28.870522 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe95052b11f nav2_behavior_tree::BehaviorTreeEngine::run()
[planner_and_controller_server-6] E0703 15:18:28.872901 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe9524c3bf8 nav2_bt_navigator::BtNavigator::bmTaskStart()
[planner_and_controller_server-6] E0703 15:18:28.875119 28373 log_comm.h:104] [planner_and_controller_server]     @     0x7fe9524f3422 nav2_util::SimpleActionServer<>::work()

facontidavide, 2024-07-03:
first, I strongly disourage using `config().blackboard->set` you must use Output/ Input Ports as explained in the tutorials.

Using the blackboard directly is an antipattern.

Second. please provide a Nav2 and possibly ROS independent way to reproduce the issue (full source code) or I will not be able to debug it.

yanfushi, 2024-07-04:
Thanks for reply, I have figured out this problem
[image]
I solved this problem by moving "behaviortree_cpp_v3/bt_factory.h" from hpp file to cpp file. But I dont know the reason. 
I will close the problem tomorrow.

closed 2024-07-05 by yanfushi