BehaviorTree/BehaviorTree.CPP #852: StatusChangeLogger is not thread safe

kfabian, 2024-07-26:
**Describe the bug**
I encountered double free memory errors in the [`RosTopicLogger` from Nav2](https://github.com/ros-navigation/navigation2/blob/4516a6a84bdce902bf9ab86f26c0e3a082113b9f/nav2_behavior_tree/include/nav2_behavior_tree/ros_topic_logger.hpp#L61) when using `ThreadedAction` nodes in my behavior tree.
I think the issue here is that the callback function of `StatusChangeLogger` gets called on each `setStatus` call and the `ThreadedAction` node [calls `setStatus` from a different thread](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/2a8a226fbbd99f524f0796e0d8a3145773c61c06/src/action_node.cpp#L204).
The concrete implementations of `StatusChangeLogger` are not aware that there implementation of `callback` might get called from different threads in parallel, so I think it would be best to prevent this from happening by adding a lock to the [`subscribeCallback`](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/2a8a226fbbd99f524f0796e0d8a3145773c61c06/include/behaviortree_cpp/loggers/abstract_logger.h#L73
).

facontidavide, 2024-09-21:
Fixed!

closed 2024-09-21 by facontidavide

kfabian, 2024-09-25:
Thanks for fixing this