BehaviorTree/BehaviorTree.CPP #853: " blackboard->get<T>(key); "caused a crash

MoYuJiang51, 2024-07-31:
I created a thread in the project to retrieve the values of specific blackboard variables at intervals of 20ms, check for changes, and log them. However, I encountered intermittent crashes without being able to identify the cause. Below are the code and stack trace.
Code:
template <typename T>
bool RobotStatusPublisher::UpdatePlanInfoKeyValue(const std::string& key, T& value) {
        auto blackboard = myBlackboard::GetInstance().GetBlackBoard();
        T    tmp_value  = blackboard->get<T>(key);
        if (value != tmp_value) {
            value = tmp_value;
            return true;
        }
        return false;
}

Crash 1: 
: bool RobotStatusPublisher::UpdatePlanInfoKeyValue<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) [clone .isra.0]
Crash 2:
 : nonstd::expected_liite::expected<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocattor<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > BT::Any::convert<std::__cxx11::basic_string<char,
std::char_traits<char>, std::allocator<char> > >(std::enable_if<std::is_same<std::_ cxx11::basic string<char,std::char_traits<char>,
std::allocator<char> >, std:: __cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >::value, void>::type*) const
 : bool RobotStatusPublisher: UpdatePlanInfoKeyValue<std::__cxx11::basic string<char, std::char_traits<char>, std::allocator<char> > >(std::__cxx11::basic_string<char, std::char_traits<char>,_std::allocator<char> > const&, std:: __cxx11::basic string<char, std::char_traits<char>, std::allocator<char> >&)
[image]

closed 2024-08-04 by MoYuJiang51

facontidavide, 2024-08-04:
Can you explain what the problem was?

MoYuJiang51, 2024-08-04:
> Can you explain what the problem was?

Here's the situation: the version I'm using is not the latest version. The function being called is 'T get(const std::string& key) const'. Inside the function, 'getAny(key);' is locked, but the return value is a pointer. I suspect that a crash occurred due to other threads modifying 'Any' during cost<T>, so I added an overloaded function to 'getAny': 'bool getAny(const std::string& key, Any& copy) const,' which returns a temporary variable. Currently, no crashes have occurred during my self-testing."
[image]