BehaviorTree/BehaviorTree.CPP #858: Bug with getInput when default parameter is passed

NelsenEW, 2024-08-19:
**Bug with getInput when default parameter is passed**

```
terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at

Thread 1 "fsm" received signal SIGABRT, Aborted.
#0  0x00007ffff769200b in raise () from /usr/lib/x86_64-linux-gnu/libc.so.6
#1  0x00007ffff7671859 in abort () from /usr/lib/x86_64-linux-gnu/libc.so.6
#2  0x00007ffff78fc8d1 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff790837c in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff79083e7 in std::terminate() ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff7908699 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff78ff33c in std::__throw_out_of_range(char const*) ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00005555555f0ddf in std::__detail::_Map_base<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::PortInfo>, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::PortInfo> >, std::__detail::_Select1st, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true>, true>::at (
    this=0x555555fadf50, __k="log_name")
    at /usr/include/c++/9/bits/hashtable_policy.h:769
#8  0x00005555555e24e3 in std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, BT::PortInfo, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::PortInfo> > >::at (this=0x555555fadf50, __k="log_name")
    at /usr/include/c++/9/bits/unordered_map.h:1007
#9  0x0000555555697602 in BT::TreeNode::getInput<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (this=0x555555fb8b40, key="log_name", 
    destination="")
    at /navigation_ws/fsm/behaviortree_cpp/include/behaviortree_cpp/tree_node.h:416
#10 0x000055555579a796 in BT::TreeNode::getInput<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (this=0x555555fb8b40, key="log_name")
    at /navigation_ws/fsm/behaviortree_cpp/include/behaviortree_cpp/tree_node.h:237
```

*How to Reproduce**
Given a node with providedPorts as shown:

```
  static BT::PortsList providedPorts()
  {
    return { BT::InputPort<std::string>("log_name", spdlog::default_logger()->name() ,"Logger name"),
             BT::InputPort<std::string>("message", "Message to be logged") };
  }
```

In the tick function, call the following:
```
  logger = spdlog::get(getInput<std::string>("log_name").value_or(spdlog::default_logger()->name()));
```

This will cause the entire thing to crash.

A quick fix is to remove the default value in the `providedPorts`:
```
  static BT::PortsList providedPorts()
  {
    return { BT::InputPort<std::string>("log_name", "Logger name"),
             BT::InputPort<std::string>("message", "Message to be logged") };
  }
```

This would no longer crash

facontidavide, 2024-08-28:
no idea...

Aglargil, 2024-09-06:
`providedPorts() `will be called in `CreateManifest()` and `BehaviorTreeFactory::registerNodeType()`, make sure you have initialized the **spdlog** before that. Or you can provide the complete code for analysis.