BehaviorTree/BehaviorTree.CPP #861: Bugprone Tick time monitoring

Petzep, 2024-08-26:
**Describe the bug**
The time measurement of a tick is bug-prone.
https://github.com/BehaviorTree/BehaviorTree.CPP/blob/2a8a226fbbd99f524f0796e0d8a3145773c61c06/src/tree_node.cpp#L109-L111 

This way of measuring time can be reordered by the compiler and will result in a time difference of 0.
A good explanation of this issue can be found here: 
[Enforcing statement order in C++](https://stackoverflow.com/questions/37786547/enforcing-statement-order-in-c/38025837#38025837)

The article also discusses a fix.

And I've also seen examples where they fix it with an atomic fence (which cost's no extra cylces)
https://stackoverflow.com/questions/22278963/how-to-prevent-timestamp-from-being-reorder-out

facontidavide, 2024-08-28:
you are welcome to submit a PR to address this.

Aglargil, 2024-09-06:
https://github.com/BehaviorTree/BehaviorTree.CPP/pull/862
This might help?