BehaviorTree/BehaviorTree.CPP #867: Building Error: Cannot open include file 'sqlite3.h': No such file or directory

jerylteo, 2024-09-17:
Using cmake version 3.30.3, on latest from Master.

Failed at step: cmake --build . --parallel

Saw a workaround at #544 to build in Release and it builds, but that forces me to only use Release mode in Visual Studios. 

Using Debug mode in VS will cause an Exception thrown when using BehaviorTreeFactory::registerNodeType<>() : Access violation reading location.

Is there a version where it works in Debug mode too? If not, is there a way to simply disable the dependencies to sqlite (and zmq since it might raise the same error)? I just need a base working version of this for me to implement in my custom engine.

jerylteo, 2024-09-17:
Using WSL on Windows 11 and repeating the commands builds successfully. For anyone that encounters the same issue.

closed 2024-09-17 by jerylteo

FlyleafPaopao, 2024-11-28:
Are there some solutions to Debug mode on windows?