BehaviorTree/BehaviorTree.CPP #880: Load external subtree from inline XML throws std::runtime_error

tropappar, 2024-10-30:
**Describe the bug**

When I include a custom subtree (called `MyTree` in this example) from an XML file, it does not work when I specify my root tree inline as string. It throws the following error:

```bash
terminate called after throwing an instance of 'std::runtime_error'
what():  Can't find a tree with name: MyTree
```

However, when I create that same subtree directly or when I include it in another tree that I load from an XML file, everything works fine.

**How to Reproduce**

This fails:
```c++
#include <behaviortree_cpp/bt_factory.h>

int main(int argc, char ** argv)
{
  BT::BehaviorTreeFactory factory;
  factory.registerBehaviorTreeFromFile("/path/to/my_tree.xml");
  std::string tree_xml =
    R"(
    <root BTCPP_format="4" >
        <BehaviorTree ID="TestTree">
            <SubTree ID="MyTree"/>
        </BehaviorTree>
    </root>)";
  auto tree = factory.createTreeFromText(tree_xml);
  auto status = tree.tickWhileRunning();
  return 0;
}
```

This works:
```c++
#include <behaviortree_cpp/bt_factory.h>

int main(int argc, char ** argv)
{
  BT::BehaviorTreeFactory factory;
  factory.registerBehaviorTreeFromFile("/path/to/my_tree.xml");
  auto tree = factory.createTree("MyTree");
  auto status = tree.tickWhileRunning();
  return 0;
}
```

Aglargil, 2024-11-01:
> **Describe the bug 描述错误**
> 
> When I include a custom subtree (called `MyTree` in this example) from an XML file, it does not work when I specify my root tree inline as string. It throws the following error:当我从 XML 文件中包含自定义子树（在本示例中称为`MyTree` ）时，当我将根树内联指定为字符串时，它不起作用。它抛出以下错误：
> 
> ```shell
> terminate called after throwing an instance of 'std::runtime_error'
> what():  Can't find a tree with name: MyTree
> ```
> 
> However, when I create that same subtree directly or when I include it in another tree that I load from an XML file, everything works fine.但是，当我直接创建相同的子树或将其包含在从 XML 文件加载的另一棵树中时，一切正常。
> 
> **How to Reproduce 如何重现**
> 
> This fails: 这失败了：
> 
> ```c++
> #include <behaviortree_cpp/bt_factory.h>
> 
> int main(int argc, char ** argv)
> {
>   BT::BehaviorTreeFactory factory;
>   factory.registerBehaviorTreeFromFile("/path/to/my_tree.xml");
>   std::string tree_xml =
>     R"(
>     <root BTCPP_format="4" >
>         <BehaviorTree ID="TestTree">
>             <SubTree ID="MyTree"/>
>         </BehaviorTree>
>     </root>)";
>   auto tree = factory.createTreeFromText(tree_xml);
>   auto status = tree.tickWhileRunning();
>   return 0;
> }
> ```
> 
> This works: 这有效：
> 
> ```c++
> #include <behaviortree_cpp/bt_factory.h>
> 
> int main(int argc, char ** argv)
> {
>   BT::BehaviorTreeFactory factory;
>   factory.registerBehaviorTreeFromFile("/path/to/my_tree.xml");
>   auto tree = factory.createTree("MyTree");
>   auto status = tree.tickWhileRunning();
>   return 0;
> }
> ```

It looks like it's because you didn't define SubTree, I tried to use your second way and still got the error, can you provide the /path/to/my_tree.xml?

tropappar, 2024-11-01:
> It looks like it's because you didn't define SubTree, I tried to use your second way and still got the error, can you provide the /path/to/my_tree.xml?

Obviously this is a toy example. A definition of MyTree could look like this:

```xml
<?xml version="1.0" encoding="UTF-8"?>
<root BTCPP_format="4">
  <BehaviorTree ID="MyTree">
    <AlwaysSuccess/>
  </BehaviorTree>
</root>
``