BehaviorTree/BehaviorTree.CPP #897: A question about Groot2

Motues, 2024-12-12:
When I import logs inside groot2, the following message appears
[image]

codes:
```c++
BT::FileLogger2 logger(tree, logFilePath);
tree.tickWhileRunning();
```
Behavior tree works fine.

facontidavide, 2024-12-13:
the code looks correct to me.

If you can provide a simple application / unit test that replicate the issue, I will look into it

Motues, 2024-12-13:
There is a simple test.
```c++
#include <behaviortree_cpp/bt_factory.h>
#include <behaviortree_cpp/action_node.h>
#include <behaviortree_cpp/condition_node.h>
#include "behaviortree_cpp/loggers/bt_file_logger_v2.h"

using namespace BT;

class SimpleAction : public SyncActionNode {
public:
    SimpleAction(const std::string& name) : SyncActionNode(name, {}) {}
    NodeStatus tick() override {
        std::cout << "SimpleAction is running..." << std::endl;
        return NodeStatus::SUCCESS;
    }
};

class SimpleCondition : public ConditionNode {
public:
    SimpleCondition(const std::string& name) : ConditionNode(name, {}) {}
    NodeStatus tick() override {
        std::cout << "SimpleCondition is checking..." << std::endl;
        return NodeStatus::SUCCESS;
    }
};

int main() {
    BehaviorTreeFactory factory;

    factory.registerNodeType<SimpleAction>("SimpleAction");
    factory.registerNodeType<SimpleCondition>("SimpleCondition");

    std::filesystem::path logFilePath = "../../Groot/Log/test.btlog";

    auto tree = factory.createTreeFromFile("../../Scripts/mytree.xml");
    FileLogger2 logger(tree, logFilePath);

    tree.tickWhileRunning();

    return 0;
}
```
The compilation environment is
* `Ubuntu24.04`
* `g++ (Ubuntu 13.2.0-23ubuntu4) 13.2.0`
* `BehaviorTree.cpp v4.6`