BehaviorTree/BehaviorTree.CPP #900: Pure virtual method called

robologics1, 2024-12-19:
Hello, the behavior tree in my system runs 24/7. However, approximately every 12 hours, I encounter a "Pure virtual method called" error, followed by a core dump. I couldn't identify any specific pattern. The error occurs in almost a different action each time. Below is the GDB backtrace output.

What would you suggest for catching this error?

Additional notes:
I am using tree_.tickExactlyOnce().
The actions consist of asynchronous actions.

`(gdb) set solib-search-path /root/ws/install_isolated/lib
(gdb) bt
#0  0x00007feda82e1e87 in raise () from /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007feda82e37f1 in abort () from /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007feda8957646 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007feda8968f06 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007feda8968f71 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007feda8969c2f in __cxa_pure_virtual () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007feda939e0d7 in BT::TreeNode::setStatus(BT::NodeStatus) ()
   from /opt/ros/melodic/lib/libbehaviortree_cpp.so
#7  0x00007feda932d9eb in std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> (), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<BT::ThreadedAction::executeTick()::{lambda()#1}> >, void> >::_M_invoke(std::_Any_data const&) () from /opt/ros/melodic/lib/libbehaviortree_cpp.so
#8  0x00007feda9d32f16 in std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>::operator()() const (this=0x7fed097f99b0) at /usr/include/c++/9/bits/std_function.h:688
#9  0x00007feda9d25c22 in std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) (this=0x7fed241e8530, 
    __f=0x7fed097f99b0, __did_set=0x7fed097f9967) at /usr/include/c++/9/future:561
#10 0x00007feda9d5c67b in std::__invoke_impl<void, void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::__invoke_memfun_deref, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&) (__f=
    @0x7fed097f9980: (void (std::__future_base::_State_baseV2::*)(std::__future_base::_State_baseV2 * const, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()> *, bool *)) 0x7feda9d25bec <std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*)>, __t=@0x7fed097f9968: 0x7fed241e8530)
    at /usr/include/c++/9/bits/invoke.h:73
#11 0x00007feda9d3e6a5 in std::__invoke<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&) (__fn=
    @0x7fed097f9980: (void (std::__future_base::_State_baseV2::*)(std::__future_base::_State_baseV2 * const, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()> *, bool *)) 0x7feda9d25bec <std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*)>) at /usr/include/c++/9/bits/invoke.h:95
#12 0x00007feda9d32c76 in void std::call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_S---Type <return> to continue, or q <return> to quit---
tate_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::once_flag&, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&)::{lambda()#1}::operator()() const (this=0x7fed097f9990) at /usr/include/c++/9/mutex:671
#13 0x00007feda9d32ca9 in void std::call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::once_flag&, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&)::{lambda()#2}::operator()() const (this=0x0) at /usr/include/c++/9/mutex:676
#14 0x00007feda9d32cba in void std::call_once<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::once_flag&, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&)::{lambda()#2}::_FUN() () at /usr/include/c++/9/mutex:676
#15 0x00007feda6f1c907 in __pthread_once_slow () from /lib/x86_64-linux-gnu/libpthread.so.0
#16 0x00007feda932e743 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<std::__future_base::_Async_state_impl<std::thread::_Invoker<std::tuple<BT::ThreadedAction::executeTick()::{lambda()#1}> >, void>::_Async_state_impl(std::tuple<BT::ThreadedAction::executeTick()::{lambda()#1}>&&)::{lambda()#1}> > >::_M_run() ()
   from /opt/ros/melodic/lib/libbehaviortree_cpp.so
#17 0x00007feda899767f in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#18 0x00007feda6f146db in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0
#19 0x00007feda83c461f in clone () from /lib/x86_64-linux-gnu/libc.so.6`

robologics1, 2024-12-23:
Additional info: The actions are being executed with the SequenceWithMemory control node. At my last test, after approximately 13 hours, the dock action is being ticked without the wait_duration returning success. The related cout logs are as follows:

[1734714319.933]: move_to_station           [33mRUNNING[0m -> [32mSUCCESS[0m
[1734714320.017]: dock                      [36mIDLE[0m -> [33mRUNNING[0m
[0m[ INFO] [1734714320.019577981, 1734714320.015523568]: Dock -> Server is up. Sending the goal.[0m
[0m[ INFO] [1734714320.216558783, 1734714320.215542560]: Dock -> Sending goal to dock_base (1/-1)[0m
[0m[ INFO] [1734714328.016282633, 1734714328.015541525]: Action succeeded[0m
[1734714328.016]: dock                      [33mRUNNING[0m -> [32mSUCCESS[0m
[1734714328.065]: wait_duration             [36mIDLE[0m -> [33mRUNNING[0m   
[0m[ INFO] [1734714328.064744878, 1734714328.060488432]: Waiting for 5.000000 seconds[0m
[1734714333.192]: dock                      [36mIDLE[0m -> [33mRUNNING[0m

At last three logs dock action turned to running before wait_duration succeed.

facontidavide, 2025-01-31:
Hi,

I will need you to provide an example to reproduce this, otherwise I will not be able to address the issue.

closed 2025-02-11 by robologics1