BehaviorTree/BehaviorTree.CPP #921: Null pointer dereference in ImportBlackboardFromJSON when entry creation fails

cktii, 2025-01-31:
## Description
A null pointer dereference occurs in `ImportBlackboardFromJSON` when attempting to use an `entry` pointer without verifying that the entry was successfully created. This happens because the code assumes `createEntry` followed by `getEntry` will always return a valid `entry`.

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/48f6c5bd1d797c73a562f1c92f404d0bb944e018/src/blackboard.cpp#L280-L295

Found in commit: https://github.com/BehaviorTree/BehaviorTree.CPP/commit/48f6c5bd1d797c73a562f1c92f404d0bb944e018

## Bug Class
Memory Safety - Null Pointer Dereference

## Root Cause

```cpp
void ImportBlackboardFromJSON(const nlohmann::json& json, Blackboard& blackboard)
{
  for(auto it = json.begin(); it != json.end(); ++it)
  {
    if(auto res = JsonExporter::get().fromJson(it.value()))
    {
      auto entry = blackboard.getEntry(it.key());
      if(!entry)
      {
        blackboard.createEntry(it.key(), res->second);
        entry = blackboard.getEntry(it.key());  // Could still return nullptr
      }
      entry->value = res->first;  // CRASH: Null pointer dereference
    }
  }
}
```

The issue occurs because:

- Code checks if initial `getEntry` returns `nullptr`
- Attempts to create `entry` if it doesn't exist
- Gets `entry` again but doesn't verify it's valid
- Dereferences the `entry` pointer without checking

### GDB

```gdb
pwndbg> p entry->value
Cannot access memory at address 0x0
pwndbg> p entry
$1 = std::shared_ptr<BT::Blackboard::Entry> (empty) = {
  get() = 0x0
}
pwndbg> p res->first
$2 = {
  _any = {
    storage = {
      dynamic = 0xffffffffe010006e,
      stack = {
        __data = "n\000\020\340\377\377\377\377\000\000\000\000\000\000\000",
        __align = {<No data fields>}
      }
    },
    vtable = 0x555556412f40 <linb::any::vtable_for_type<long>()::table>
  },
  _original_type = {
    _M_target = 0x7ffff7c6e610 <typeinfo for long>
  }
}
pwndbg>
```

## Stack trace

```
#0  0x00005555558f5d1e in BT::ImportBlackboardFromJSON (json=..., blackboard=...)
```

## Impact

Impact

- Program crash on malformed input
- Potential security vulnerability if JSON input is untrusted
- Could be used for denial of service attacks

facontidavide, 2025-01-31:
Thanks for the accurate description.

The more interesting question here is why "blackboard.createEntry(it.key(), res->second);" is not doing what is expected to do.

Do you have perhaps a simple example to reproduce the issue?

cktii, 2025-01-31:
I must have a reproducer lying around that I used when debugging this. I’ll try to attach one ASAP. 

While I’m at it. I’ll try to deliver a minimal PoC for all the other ones as well!

cktii, 2025-02-03:
Okay, after some debugging I came to the conclusion that unlike stated above in the original description suggested this was a simple null pointer dereference from `createEntry`/`getEntry`, this seems to be a more complex issue involving the `any` type system.

## Minimal Reproducer
```cpp
#include "behaviortree_cpp/blackboard.h"
#include <iostream>
#include <nlohmann/json.hpp>

int main() {
    try {
        auto bb = BT::Blackboard::create();
        
        // Set initial value 
        bb->set("mmmmmm5CnCaaG", -535822226);
        
        // Create JSON that will trigger the crash
        std::string json_str = R"({
            "@@@@@@@@@@@@@@@": -535822226,
            "mmmmmm5CnCaaG": null
        })";
        
        // Parse and import JSON - will crash
        auto json = nlohmann::json::parse(json_str);
        BT::ImportBlackboardFromJSON(json, *bb);
    }
    catch(const std::exception& e) {
        std::cerr << "Exception: " << e.what() << std::endl;
        return 1;
    }
    return 0;
}
```

## Updated stack-trace

```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==3669987==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000010
    #0 0x7dfae074647f in linb::any::swap(linb::any&) /home/user/BehaviorTree.CPP/include/behaviortree_cpp/contrib/any.hpp:183:32
    #1 0x7dfae074647f in linb::any::operator=(linb::any const&) /home/user/BehaviorTree.CPP/include/behaviortree_cpp/contrib/any.hpp:129:18
    #2 0x7dfae074647f in BT::Any::operator=(BT::Any const&) /home/user/BehaviorTree.CPP/include/behaviortree_cpp/utils/safe_any.hpp:306:14
    #3 0x7dfae074647f in BT::ImportBlackboardFromJSON(...) /home/user/BehaviorTree.CPP/src/blackboard.cpp:292:20
```

The issue appears to be related to type handling in the Any implementation when dealing with null values during JSON deserialization. This suggests the fix might need to handle:

- Proper null value handling in the JSON import
- Type safety checks in the Any implementation
- Validation of entry state before assignment

Aglargil, 2025-02-06:
```cpp
#include "behaviortree_cpp/blackboard.h"
#include <iostream>
#include <nlohmann/json.hpp>

int main() {
    try {
        auto bb = BT::Blackboard::create();
        
        // Set initial value 
        bb->set("mmmmmm5CnCaaG", -535822226);
        
        // Create JSON that will trigger the crash
        std::string json_str = R"({
            "@@@@@@@@@@@@@@@": -535822226,
            "mmmmmm5CnCaaG": null
        })";
        
        // Parse and import JSON - will crash
        auto json = nlohmann::json::parse(json_str);
        BT::ImportBlackboardFromJSON(json, *bb);
    }
    catch(const std::exception& e) {
        std::cerr << "Exception: " << e.what() << std::endl;
        return 1;
    }
    return 0;
}
```

Hi, @facontidavide 

The reason why `blackboard.createEntry(it.key(), res->second); `did not perform as expected in this case is: When the current key was "@@@@@@@@@@@@@@@" (15 @ symbols), since the key starts with @, `rootBlackboard()->createEntryImpl(key.substr(1, key.size() - 1), info); `created an entry with key "@@@@@@@@@@@@@@" (14 @ symbols). The critical issue was that the implementation didn't recursively call `createEntry` during this process, while the `getEntry` method does employ recursive calls to `getEntry`. This inconsistency ultimately led to failing to find the entry when searching for an empty string key (after recursively stripping all @ prefixes). 

I've submitted a PR to fix this behavior #929

facontidavide, 2025-02-09:
fixed making that syntax illegal, as it should be

facontidavide, 2025-02-09:
thanks

[image]

closed 2025-02-09 by facontidavide