BehaviorTree/BehaviorTree.CPP #931: Segmentation fault when Node not recognized

kgilarski, 2025-02-07:
There is a segmentation fault when using createTreeFromText with text containing unrecognized (unregistered) node. 

The problem is in the current master branch. Branch 4.6.2 throws in such a case an exception: 
Error at line X: -> Node not recognized: Y

Backtrace:
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007fa9b9dd1aaa in BT::VerifyXML(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, BT::NodeType, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::NodeType> > > const&)::{lambda(tinyxml2::XMLElement const*)#2}::operator()(tinyxml2::XMLElement const*) const () from /usr/local/lib/libbehaviortree_cpp.so
[Current thread is 1 (Thread 0x7fa9b3fff6c0 (LWP 10))]
#0  0x00007fa9b9dd1aaa in BT::VerifyXML(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, BT::NodeType, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::NodeType> > > const&)::{lambda(tinyxml2::XMLElement const*)#2}::operator()(tinyxml2::XMLElement const*) const () from /usr/local/lib/libbehaviortree_cpp.so
#1  0x00007fa9b9dd1670 in BT::VerifyXML(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, BT::NodeType, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::NodeType> > > const&)::{lambda(tinyxml2::XMLElement const*)#2}::operator()(tinyxml2::XMLElement const*) const () from /usr/local/lib/libbehaviortree_cpp.so
#2  0x00007fa9b9dd1670 in BT::VerifyXML(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, BT::NodeType, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::NodeType> > > const&)::{lambda(tinyxml2::XMLElement const*)#2}::operator()(tinyxml2::XMLElement const*) const () from /usr/local/lib/libbehaviortree_cpp.so
#3  0x00007fa9b9dd79ac in BT::VerifyXML(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, BT::NodeType, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, BT::NodeType> > > const&) () from /usr/local/lib/libbehaviortree_cpp.so
#4  0x00007fa9b9dd82a3 in BT::XMLParser::PImpl::loadDocImpl(tinyxml2::XMLDocument*, bool) () from /usr/local/lib/libbehaviortree_cpp.so
#5  0x00007fa9b9d673ce in BT::BehaviorTreeFactory::createTreeFromText(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<BT::Blackboard>) () from /usr/local/lib/libbehaviortree_cpp.so
```

facontidavide, 2025-02-07:
Thanks. Do you have an example that can reproduce the error?

kgilarski, 2025-02-08:
I can't provide an exact example, but I have prepared something which might help.

```
void execute(const string &xml) {
        BehaviorTreeFactory factory;
        auto tree_ptr = std::make_shared<BT::Tree>( factory.createTreeFromText( xml ));
}
```

where xml is as follows:
[FED801.txt](https://github.com/user-attachments/files/18718636/FED801.txt)

The factory has registered 21 nodes using registerNodeType method but non of them is in the xml file. 
The whole example is a situation when somebody delivers incorrect tree by mistake and we want to get an exception, not a segmentation fault.

facontidavide, 2025-02-09:
i added a unit test, but i am not able to reproduce the error... https://github.com/BehaviorTree/BehaviorTree.CPP/commit/27939021acfb4738843b435664077a1cf4c3fbf9

kgilarski, 2025-02-10:
Maybe somebody fixed it in the meantime. Segmentation fault is in version 2025-02-05. In version 2025-02-09 there is an exception.

Below is the code I used to reproduce the problem.

```
#include "behaviortree_cpp/bt_factory.h"
#include <memory>
#include <iostream>
#include <fstream>
#include <sstream>
#include <string>

int main() {

    std::ifstream file("../../resource/FED801.txt");  // Open the file
    if (!file) {  // Check if file opened successfully
        std::cerr << "Unable to open file\n";
        return 1;
    }

    std::stringstream buffer;
    buffer << file.rdbuf();  // Read the contents of the file into the stringstream

    std::string fileContents = buffer.str();  // Convert the stringstream to string
    std::cout << "File Contents:\n" << fileContents << std::endl;

    file.close();  // Close the file

    try {
        BT::BehaviorTreeFactory factory;
        auto tree_ptr = std::make_shared<BT::Tree>(factory.createTreeFromText(fileContents));
    } catch (std::exception & ex) {
        std::cout << "Exception: " << ex.what() << std::endl;
    }
}
```

Result depends on the BehaviourTree.CPP version:

	2024-02-19
		commit 3073f2904921fa248185f293f046358c97a42eed (HEAD -> master, origin/master, origin/HEAD)
		Author: Davide Faconti <davide.faconti@gmail.com>
		Date:   Mon Feb 19 13:20:52 2024 +0100

		Exception: Error at line 5: -> Node not recognized: Is431
		
	2025-02-05
		commit 370133ebd994b75f914c9bd0bda3e203777e70df (HEAD)
		Author: Davide Faconti <davide.faconti@gmail.com>
		Date:   Wed Feb 5 01:08:48 2025 +0100
		
		Naruszenie ochrony pamięci (zrzut pamięci) - Polish Segmentation Fault                
		
	2025-02-09
		commit 87d2f03ea2828908402864a1df4fd61e689b4386 (HEAD -> master, origin/master, origin/HEAD)
		Author: Davide Faconti <davide.faconti@gmail.com>
		Date:   Sun Feb 9 21:43:47 2025 +0100

		Exception: Error at line 5: -> Unknown node type: Is431

closed 2025-02-12 by facontidavide